atmelcorp/atmel-software-package #1: Problems with NAND flash applet

thibef, 2016-05-03:
Hi. 
When trying to use the nand flash applet with the SAMA5d2 xplained ultra board i get the following error:
Error: Applet AT25/AT26 Nand Flash failed to initialize (status=15)

Since no configuration file for sam-ba was provided i just tried to adapt the configuration for the serial flash applet in the SAMA5D.qml that configures the applets to be used with SAM-BA 3.0

I can use the serial flash applet just fine but not the nand flash one.

Has anyone found the same problem?

loiclefort, 2016-05-03:
Hi,
The NAND applet is fully supported starting with SAM-BA 3.1.1 (released only a few days ago).
But the SAMA5D2 Xplained Ultra board does not have a NAND, so it is expected that the NAND applet would not detect anything.

thibef, 2016-05-03:
Hi. 
Thanks for your help. So I should use the SDMMC applet?

thibef, 2016-05-03:
In case the SDMMC applet is the right one, is there some documentation somewhere describing how should I add this applet to the SAMA5D2.qml file?

Thanks

loiclefort, 2016-05-03:
Yes, on the SAMA5D2 Xplained Ultra you can boot on SD/MMC card or on the on-board eMMC.
Just download the 3.1.1 version on Atmel website, it contains examples for both.

closed 2016-05-04 by loiclefort