atmelcorp/atmel-software-package #3: About the binary generated from IAR

bbucka, 2016-08-11:
From software package 2.2, the binaries generated from IAR doesn't work on SAMA5D2-xplained Ultra evaluation board. The version of my IAR is 7.7. I think it has something to do with BSS section.

lfazio, 2016-08-12:
We need more information to help you on this specific problem:
- Is there one specific example or all?
- What VARIANT do you use?

Have you tested it again with the latest 2.4 release?

bbucka, 2016-08-12:
I downloaded 2.4 release.
I used the example of getting-started.
I am using cywin64, gnu make-3.81, coreutils-5.3.0 and GNU Tools ARM Embedded 5.3 2016q1 on windows 10.
I built like "$ make TARGET=sama5d2-xplained".
I got "getting-started_sama5d2-xplained_sram.bin" with the size of 51KB.
I downloaded it into the serial flash. I reset the board and it worked.
I made the workspace for IAR like "$ make TARGET=sama5d2-xplained iar".
I compiled in IAR 7.7 and got "getting-started_sama5d2-xplained_sram.bin" with the size of 67KB.
I downloaded it into the serial flash. I reset the board and it didn't work.
I just got the message "RomBOOT".

loiclefort, 2016-08-12:
I have tested with IAR 7.6. Binary is 39KB and boots fine.
I will upgrade to 7.7 and test again.

loiclefort, 2016-08-12:
Confirmed with IAR 7.7, got a 67KB binary that doesn't boot, and this is expected since the SAMA5D2 ROM-code will only boot binaries <64KB.
Need to investigate on the size difference between IAR 7.6 and 7.7.

patvilchez, 2016-08-12:
If you get 67KB it can not boot.
The limit is 64KB.

bbucka, 2016-08-12:
If I use "qspi0" for VARIANT and QSPI XIP mode, the size of the binary wouldn't matter?

loiclefort, 2016-08-12:
Yes, in QSPI XIP mode the size of the binary is only limited by the size of the QSPI flash.

closed 2016-08-16 by bbucka

loiclefort, 2016-08-16:
@bbucka here is a workaround to generate correct binaries using IAR 7.7:
in target/sama5d2/toolchain/iar/sram.icf, remove the line
`define block BSS with alignment = 32 { zi }`
and change line
`place in RAM_region { block BSS };`
to
`place in RAM_region { zi };`