atmelcorp/atmel-software-package #4: the size of the "usb_mass_storage" example

bbucka, 2016-08-16:
I built the example of "usb_mass_storage" from 2.4 with gnu make in cygwin.
The size of  "usb_mass_storage_sama5d2-xplained_ddram.bin" is about 8.3MB.
Is it the right size?

loiclefort, 2016-08-17:
Thanks for the report. There was indeed an issue in the ddram linker script.
It will be fixed in next release. In the mean time, you can edit the linker script: In `target/sama5d2/toolchain/gnu/ddram.ld`, move section `region_cache_aligned_const` just before region `region_ddr`. This puts all `NOLOAD` sections together at then end of the binary so the linker can omit them in the output file.

bbucka, 2016-08-17:
I have a quick question. What is the bootstrap.c used for?

loiclefort, 2016-08-17:
It is used when debugging a program built to run from DDRAM using GDB.
The GDB script will first load the bootstrap program in SRAM and run it to setup clocks & initialize DDR, then it will load the "real" program in DDR and run it.

closed 2016-08-18 by bbucka