atmelcorp/atmel-software-package #6: SAMA5D2 nandflash problems with sam-ba 3.1.3

sepldj, 2016-08-31:
Reproducing question posted to at91sam forum:

> I'm facing some issues working with the nandflash applet provided with sam-ba version 3.1.3. I'm using the examples/sama5d2/nandflash/nandflash-usb.qml script.
> 
> When invoked, the applet fails with the following error:
> 
> Applet.qml:95: Error: Could not initialize applet (status: undefined)
> 
> To investigate the issue further, I started hacking into the applet source (https://github.com/atmelcorp/atmel-soft ... ash/main.c):
> 
> I don't have any jtag interfaces at my disposal, so the debugging is quite slow and painful, however what I find is, the issue appears to be related to some sort of RAM/stack corruption.
> 
> The reason for the above error is, the applet crashes during initialisation in handle_cmd_initialize() - code never returns from applet_set_init_params().
> 
> One of the debug steps is, I've stubbed out the following functions to return success.
> 
> ```
> handle_cmd_read_info()
> handle_cmd_erase_pages()
> handle_cmd_read_pages()
> handle_cmd_write_pages()
> ```
> 
> At this point the initialisation completes successfully (see output snippet below):
> 
> ```
> Opening serial port 'COM6'
> Connection opened.
> XXXXXXXXXXXX
> args: 0,5,2,8,0
> Detected memory size is 536870912 bytes.
> XXXXXXXXXXXX
> Erased 0 bytes at address 0x00000000 (0.00%)
> Erased 0 bytes at address 0x00000000 (0.00%)
> Erased 0 bytes at address 0x00000000 (0.00%)
> Erased 0 bytes at address 0x00000000 (0.00%)
> Erased 0 bytes at address 0x00000000 (0.00%)
> ...```

I've since got the HW guys to connect the jtag port and got the IAR debugger going. This leads me to the next question:

If I download the applet via IAR - jtag, I get the debugger, but I don't appear to have the environment setup by SAM-BA (the applet executes and returns to la-la land - presumably because the applet wasn't invoked by the bootROM loader?).

If I use SAM-BA to download the applet, I loose the ability to debug it.

So the question is, what's the best way to get the best of both worlds? I'd like to be able to invoke the applet from the SAM-BA command line application, but still have the ability to debug it.

Thanks in advance.

loiclefort, 2016-08-31:
Debugging the applets using IAR is not yet implemented, only GDB is supported:

    cd samba_applets/nandflash
    make TARGET=sama5d2-generic debug

The GDB script provided contains a "runapplet" macro:

    runapplet 0 0 5 2 8 0

You can also get some information on the device console. Applets will display some status on the UART configured in the boot configuration.

Can you provide your NAND model? The cause of the issue could be that the memory is not ONFI compliant. We don't yet support non-ONFI NANDs.

sepldj, 2016-08-31:
Thanks for getting back so quickly.

1. Ok I'll install gdb and try to debug the applet.
2. I don't even have a console at this time (see my original post as to where the problem occurs)
3. The flash device is  S34MS04G100BHI000 which is ONFI compliant (http://www.cypress.com/file/207116/download). Again as the log output in the original post shows, the device data was detected by the applet and hence SAM-BA.

Thanks again.

sepldj, 2016-09-02:
Have some further info:

1. Applets load and execute correctly if I use the jtag interface i.e. the nandflash-jlink.qml script from [this location](https://github.com/atmelcorp/sam-ba/tree/14e039927fcd7bd0e95a079093615bb1e90e6360/examples/sama5d2/nandflash).
2. Applet crashes during initialisation when I use the USB interface - nandflash-usb.qml

From SAM_BA's point of view, are the two interfaces identical or are there sufficient differences to explain this inconsistency?

loiclefort, 2016-09-02:
There should be no difference between JTAG and USB.
Do you observe the problem both with the applets provided in the SAM-BA package and with the applets you have rebuilt?

sepldj, 2016-09-02:
OK the problem appears to be with the SAM-BA utility, The nandflash applet is ~36K (which happens to be larger than a magic number MAX_BUFFER_SIZE = 32 * 1024 in the SAM-BA utility). We increased the buffer to 64K and the applet loaded and executed without an issue. I imagine you'll want to close this ticket and open a separate one in the SAM-BA repo?

loiclefort, 2016-09-02:
Yes, can you please open an issue on the SAM-BA repo?

closed 2016-09-05 by sepldj