atmelcorp/atmel-software-package #7: Problems with QSPI-XIP

dandersonee, 2016-09-01:
I am using ATSAMA5D2B-XULT with SAMA5D27 Rev B silicon.
I built the getting_started example for configuration QSPI0 using IAR 7.70.1
When launched from the debugger, the example runs properly.
When I try to reboot and let RomBOOT run the program, it does not launch, it just prints RomBOOT.
I have used sam-ba to set QSPI0_IOSET3 and QSPI_XIP_MODE.

BSCR=0x00000004 / BUREG0,VALID
BUREG0=0x00240ffe / QSPI0_IOSET3,QSPI1_DISABLED,SPI0_DISABLED,SPI1_DISABLED,NFC_DISABLED,SDMMC0_DISABLED,SDMMC1_DISABLED,UART1_IOSET1,JTAG_IOSET1,EXT_MEM_BOOT,QSPI_XIP_MODE
RomBOOT version is:
>
v1.2 Dec  1 2015 09:20:36
>

I discovered that the cause is that RomBOOT must check the sixth vector at address 0xD0000014 (normally image size) and saw that the value was >64K and therefore rejected QSPI0 as a boot source.

I solved it by fixing \target\sama5d2\toolchain\iar\cstartup.s
In the section:
        /* Dummy vector table for ROM-code for cases when the real vector table
         * is relocated (QSPI-XIP) */
I changed the sixth line
FROM:        ldr     pc, =__iar_program_start
TO:        DCD 0

loiclefort, 2016-09-01:
Confirmed, thanks.
A fix will be integrated in next release.

dandersonee, 2016-09-01:
This same issue impacts the qspi_xip example which is the other example I was trying to use to test QSPI_XIP.  That example programs a hex table xip_program[] of the getting_started example in QSPI Flash.
The file examples\qspi_xip\getting-started_sama5d2-xplained_qspi0.h also needs to have the 6th vector changed to 0x00000000 in this file.

loiclefort, 2016-09-02:
Yes, good catch. We will rebuild getting-started_sama5d2-xplained_qspi0.h before next release.
Thanks

closed 2016-09-07 by loiclefort

dandersonee, 2016-09-21:
The file examples\qspi_xip\getting-started_sama5d2-xplained_qspi0.h also needs to have the 6th vector changed to 0x00000000 in this file. (comment above).
This file was NOT rebuilt in the v2.5 release.
Reopen issue or should I submit a new issue?

loiclefort, 2016-09-22:
Sorry, I forgot to do it before the release.
I have pushed a fix on master branch.

xzp114, 2017-04-12:
Hi,
I built the qspi_xip example for configuration sram using IAR 7.60.2 in debugger mode, my board can run properly. When i flash the bin code(configuration qspi0) and reboot, it doesnot work and prints RomBOOT? Why? I have used QSPI_XIP_MODE!

loiclefort, 2017-04-14:
Hi @xzp114 
I'm not sure of what you are trying to achieve here... The qspi_xip examples shows how to setup the QSPI XIP in mode (also called continuous read mode) and jump to run code directly from the QSPI. It is *not* an example of how to boot directly in QSPI XIP mode. If you want to boot directly from QSPI in XIP mode, you can use any *other* example and build it for "qspi0" variant.
Also, do to not hijack issues like this, please create a new issue instead.