atmelcorp/atmel-software-package #8: nandflash applet returns an error when reading back written content

sepldj, 2016-09-02:
We're using a 512M NAND flash with an Open NAND Flash Interface (ONFI) 1.0 compliant interface.

1. The nandflash applet successfully initialises and works out a default PMECC configuration: 0xc0080405. (This I got by passing a 0 as the header when invoking the init command).
2. I'm able to download a binary image without any errors using the QML scripts shipped with SAM-BA 3.1.3
3. When reading back the flash content using the QML scripts, I get a read failure. The console output from the applet indicates that it is an error generated by the PMECC: ```-E- ecc_read_page_with_pmecc: at B0.P0 Unrecoverable data```
4. If I program the binary with the PMECC turned off (i.e. setting the PMECC configuration value to 0xc0080404), the read completes successfully.
5. Results are repeatable and identical across two boards

This suggests the ECC data written along with the binary data is not as expected by the read function. I'm not quite able to work out what configuration parameters I need to use . Are you able to provide any ideas?

The device is a S34MS04G1 by Cypress with the following specs ([data sheet](http://www.cypress.com/file/207116/download)):

**Density**
- 1 Gbit / 2 Gbit / 4 Gbit

**Architecture**

- Input / Output Bus Width: 8-bits / 16-bits
- Page Size:

  - x8 = 2112 (2048 + 64) bytes; 64 bytes is spare area
  - x16 = 1056 (1024 + 32) words; 32 words is spare area

- Block Size: 64 Pages

 - x8 = 128k + 4k bytes
 - x16 = 64k + 2k words

- Plane Size:

 - 1 Gbit / 2 Gbit: 1024 Blocks per Plane

  - x8 = 128M + 4M bytes
  - x16 = 64M + 2M words

 - 4 Gbit: 2048 Blocks per Plane

  - x8 = 256M + 8M bytes
  - x16 = 128M + 4M words

- Device Size:

 - 1 Gbit: 1 Plane per Device or 128 Mbyte
 - 2 Gbit: 2 Planes per Device or 256 Mbyte
 - 4 Gbit: 2 Planes per Device or 512 Mbyte

**NAND Flash Interface**

- Open NAND Flash Interface (ONFI) 1.0 compliant
- Address, Data and Commands multiplexed

**Electronic Signature**

 – Manufacturer ID: 01h

loiclefort, 2016-09-02:
Can you try with 0xc0082405 ?

sepldj, 2016-09-05:
Yes that works. So just to get my head around it, why 4 bits for ECC instead of 2? Could you point out any docs that describe how to decide these params please?

loiclefort, 2016-09-05:
From the NAND datasheet:
- page size: 2048 bytes (+ 64 spare)
- sector size: 512 bytes (+ 16 spare)
- 1-bit ECC per 528 bytes

-> 4 sectors per page -> 4-bit ECC per page

I will check the NAND applet, it should have provided the correct value automatically.

loiclefort, 2016-09-06:
I have confirmed that there was a bug in the applet.
The ONFI table provides the minimum ECC correction required for 512 bytes, and the applet did not scale this value to the actual page size. A fix will be provided in next release.

sepldj, 2016-09-06:
Cool. Thanks for you assistance

closed 2016-09-07 by loiclefort