atmelcorp/atmel-software-package #10: PIT generates spurious interrupts when AIC Protect mode is enabled

dandersonee, 2016-09-12:
I have generated a simple test case to show the problem.
It counts spurious interrupts over a 1 second interval both without AIC protect mode enabled and with AIC protect mode enabled.
This example runs on the ATSAMA5D2B-XULT with VARIANT=SRAM
I tested using IAR EWARM 7.70.1
Put the attached pit_test in the examples directory of the Atmel Software Package.
NOTE: This requires the patched Atmel Software Package per Issue: https://github.com/atmelcorp/atmel-software-package/issues/9
Can you see something I have configured wrong or is this an undocumented errata in the chip?

The only workaround I have found is to add Preprocessor defined symbol CONFIG_TIMER_POLLING which avoids using wfi sleeping during the wait and generates no spurious interrupts.

Chip: SAMA5D27 Rev B
Board: ATSAMA5D2B-XULT

Console Output:
-- PIT Test Example --
Softpack v2.4
Built for sama5d2-xult
Processor: SAMA5D27-CU
Processor clock: 498 MHz
Master clock: 166 MHz
MMU is disabled
I-Cache is disabled
D-Cache is disabled

Spur count without AIC PROT=0
Spur count with AIC PROT=1000

[pit_test.zip](https://github.com/atmelcorp/atmel-software-package/files/468251/pit_test.zip)

dandersonee, 2016-10-03:
Have not seen any updates for this.  Is someone at Atmel looking into this problem?  Working toward publishing chip errata?  Considering fixing for Rev C silicon?  Wondering if I needed to submit through technical support request or elsewhere to make the right people aware if this is not a driver bug.

lfazio, 2016-10-06:
Dear @dandersonee, the problem has been reported. You may anyway submit a technical support request.

dandersonee, 2016-10-17:
Just additional info.
Confirmed problem also in Atmel Software Package 2.5.1
I tried without CONFIG_TIMER_POLLING by commenting out the call to irq_wait from timer.c-timer_wait() to just skip WFI sleeping in between polling PIT status but still got spurious interrupts.

javier-box, 2016-10-17:
The timer module should not use the PIT to begin with. 

Many RTOS implement it as it's tick source, and will clearly conflict with the RTOS. Instead a TC should be used for the timer module. This is what I had to do finally, reimplement the timer module to usese TC and not the PIT. Also, resolution of timer_configure is in usec not nano as said in the comments.

This brings up the subject of the tc.c. tc_find_best_clock_source does not find the best frequency for the given freq. It always gives the slow clock.

On a different topic, please quit using protected as struct field names, it is a C++ keyword.

dandersonee, 2016-10-18:
@javier-box All valid points but they should be written up as separate bug reports, not as comments to this specific issue so they can be addressed.

lfazio, 2016-10-18:
@dandersonee spurious irq still under investigation.
@javier-box In the next release, the PIT has already been replaced by a TC and work is still on going on TC.

We are moving step by step and we try to improve the features as well as the quality of the softpack.

javier-box, 2016-10-18:
@dandersonee, will do next time.
@lfazio, excellent. Looking forward to the next release.

dandersonee, 2016-10-31:
@lfazio, any updates on this investigation?
Your previous comment sounded like you were also changing the timer utility to use TC instead of PIT.  Hope this avoids the issues of reprogramming PIT in sdmmc and libsdmmc as a high resolution timer for short delays which messes up using it as a 1ms tick for the RTOS.
Is a new library release imminent?

lfazio, 2016-11-03:
I'm back from vacations and I got a workaround for this problem. It has been reproduced under a simulation environment and explained.

Fix and explaination in 007e8e1ea076fd73e1272a9fca9c616d74abd4f3.

We plan a new release soon, probably next week.

closed 2016-11-03 by lfazio

dandersonee, 2016-11-04:
@lfazio: Glad you verified the issue and have a software workaround for the problem.
I assume this problem affects both the AIC and the SAIC.
Regarding the checkin implementing the workaround:
In cstartup.s for sama5d2 for IAR you fixed fiqHandler but not irqHandler. (This is different than the next 3)
In cstartup.S for sama5d2 for GNU you fixed irqHandler but not fiqHandler.
In cstartup.s for sama5d4 for IAR you fixed irqHandler but not fiqHandler.
In cstartup.S for sama5d4 for GNU you fixed irqHandler but not fiqHandler.
sama5d3 and sam9xx5 dont use fiqHandler so those are OK.

dandersonee, 2016-11-04:
When I made the missing edits in irqHandler to cstartup.s for sama5d2 for IAR and ran the pit_test example I had submitted to demonstrate the problem, it did show 0 spurious interrupts with AIC protect enabled so that is good news about the workaround.

lfazio, 2016-11-04:
Too quick commit after vacations... I amend it in a new commit on top of master.

Commit: f90e27f26d56bc74676dc0250048830a8aeecca8