atmelcorp/atmel-software-package #12: using RTC in UTC mode

bbucka, 2016-09-19:
The example for RTC in the software package is using only Gregorian mode.
Is there any example for RTC using UTC mode?
I tried to use UTC mode, but I couldn't.
I tried as below 
                    uint32_t ltime = 0;
                    RTC->RTC_WPMR = 0x525443;
                    rtc_set_hour_mode(4);
                    RTC->RTC_CR |= RTC_CR_UPDTIM;
                    while ((RTC->RTC_SR & RTC_SR_ACKUPD) != RTC_SR_ACKUPD) ;
                    RTC->RTC_SCCR = RTC_SCCR_ACKCLR;
                    RTC->RTC_TIMR = 0;
                    RTC->RTC_CR &= (uint32_t) (~RTC_CR_UPDTIM);
                    RTC->RTC_SCCR |= RTC_SCCR_SECCLR;
After that, if I read the register of RTC->RTC_TIMR, it always gave 0x4100000.
I can't figure out what the value is.
I don't think that is the number of seconds elapsed since the reference time(1972.6.30).
My MPU is SAMA5d27.
Thank you.

loiclefort, 2016-09-19:
We don't have an example for UTC mode at the moment.
Not sure it this will help you, but:

- You are writing the wrong value in RTC_WPMR. Should be:
RTC->RTC_WPMR = 0x525443 **<< 8**;
or even better:
RTC->RTC_WPMR = RTC_WPMR_WPKEY_PASSWD;

- The datasheet indicates that you should start the update just after a second periodic event ("25.5.6.2 UTC Mode": "The update of the UTC time field must be synchronized on a second periodic event by either polling the RTC_SR.SEC status bit or by enabling the SECEN interrupt in the RTC_IER.")

lfazio, 2016-09-29:
@bbucka Can we close this issue?

bbucka, 2016-09-29:
Do you have any plan to implement the UTC mode for RTC in the future?

closed 2016-09-29 by bbucka

lfazio, 2016-09-29:
I can't give a deadline, but your request is now in the softpack backlog.

If you want to contribute, feel free to implement it on your side and send a pull request. We will review carefully your contribution.