atmelcorp/atmel-software-package #15: About QSPI Driver

bbucka, 2016-09-29:
I am trying to download a firmware into a qspi flash(MX25L3206) using sam-ba_3.1.3.
Downloading fails printing out "Error while detecting QSPI flash chip".
The driver recognizes my flash as "MX25L3205".

The point generating errors is as below:
static bool _macronix_set_dummy_cycles(struct _qspiflash *flash,
{
...
	if ((cr & mask) != val)
		return false;
} 

I can download if I remove the part above and DMA operation.
However the ROM code inside the MCU just prints out "RomBOOT".
Is the ROM code using the same functions to identify the flash?

I am using SAMA5d21 for MCU and QSPI1 IOSET2 for QSPI.
I am using appletWriteBootCfg(BootCfg.BUREG0, BCW.fromText("EXT_MEM_BOOT,UART3_IOSET3,JTAG_IOSET1," + "SDMMC1_DISABLED,SDMMC0_DISABLED,NFC_DISABLED," + "SPI1_DISABLED,SPI0_DISABLED," + "QSPI0_DISABLED,QSPI1_IOSET2"))  for boot config.

lfazio, 2016-09-30:
sam-ba 3.1.4 publication is on going. When it is ready, you may test with this latest release.

CyrillePitchen, 2016-10-03:
Hi bbucka,

according to the MX25L3206 datasheet, this memory has no Configuration Register. Hence the Read Configuration Register (15h) op code is not recognized.

ROM Code v1.1 and v1.2 share code with sam-ba applet to manage qspi memories so the very same
test is performed by those ROM Codes: they won't be able to boot from your MX25L3206 memory.

ROM Code v1.3 uses SFDP tables (Serial Flash Discoverable Parameter) to dynamically discover the memory settings (op codes, settings, ...). Also The ROM Code v1.3 no longer tries to set the number of dummy cycles for Fast Read commands but instead it assumes the memory still uses its factory settings, which are provides by the SFDP tables. When SFDP tables are not provides, the ROM Code uses hard coded settings as a fallback, which depends on the manufacturer. In this case also, the ROM Code no longer tries to modify the number of dummy cycles.

Since ROM Code v1.3 is not available yet, I suggest you to use some other memory such as one from the MX25L{128,256,512}73g series, like the one embedded on the sama5d2 xplained rev B.

bbucka, 2016-10-03:
Thank you for your reply. I am going to switch to a SPI flash which is supported. Thank you again.

closed 2016-10-03 by bbucka