atmelcorp/atmel-software-package #20: SAMA5D2 Xplained 18 bit LCD pin def are wrong

javier-box, 2016-11-06:
Maybe someone should run through the SAMA5D2 pin def before releasing.

This one issue cost me money. This software pack should really be an alpha version. I think I'll put the D2 aside for a few years to let it cook somemore. It is not ready.

dandersonee, 2016-11-07:
@javier-box : are you referring to the issue I already found [https://github.com/atmelcorp/atmel-software-package/issues/14](url) or have you found something else?
I am using the SAMA5D2B-XULT eval board with the PDA TM4301B display.  Appears to be working so far.

javier-box, 2016-11-08:
yes that is one, but there is something else wrong. if you draw a gradient the oder of the colors is wrong. almost as if there is a nibble reversed or it only 4 bits work.

loiclefort, 2016-11-09:
@javier-box we need more information in order to be able to help you: which board and display are you using? are you using the latest software package from git?

javier-box, 2016-11-09:
SAMA5D2 and a 800x480 24 bit TFT

javier-box, 2016-11-09:
one more thing. I have the same identical wiring of the 50 pin zif working perfectly on a SAMA5D3 and these are supposed to be compatible, except one is upper  18 bits and the other 24. Best way to test colors is with a gradient for each RGB byte.

loiclefort, 2016-11-09:
@javier-box I have tested again here with the same setup as @dandersonee (SAMA5D2-XULT with Rev B chip, PDA 4.3" display). Everything seems to be working fine, gradients look good.

loiclefort, 2016-12-07:
@javier-box did you manage to get your LCD to work?
Without more information from you, we will close this issue at the end of the week since we can't reproduce the problem.

closed 2016-12-10 by loiclefort