atmelcorp/atmel-software-package #22: Pmc.c-pmc_get_plla_clock() and pmc_get_main_clock() return incorrect values.

dandersonee, 2016-12-06:
A coding error calculates frequency based on the wrong oscillator value.
The pmc_get_plla_clock() function has:
	if (PMC->CKGR_MOR & CKGR_MOR_MOSCSEL)
		pllaclk = _pmc_oscillators.internal.main; /* on-chip main clock RC */
	else
		pllaclk = _pmc_oscillators.external.main; /* external crystal */
but a 0 for the CKG_MOR_MOSCSEL signifies the internal 12MHz RC oscillator and a 1 signifies the external 8 to 24 MHz crystal.

The pmc_get_main_clock() function also has the logic backwards:
uint32_t pmc_get_main_clock(void)
{
	if (PMC->CKGR_MOR & CKGR_MOR_MOSCSEL)
		return _pmc_oscillators.internal.main; /* on-chip main clock RC */
	else
		return _pmc_oscillators.external.main; /* external crystal */
}

Since demo board used a 12 MHz external oscillator that matches the internal 12 MHz RC oscillator  the swap was not noticed.  When custom board used a 24 MHz crystal, the problem was discovered.

loiclefort, 2016-12-06:
Thanks for noticing this, it will be fixed for next release.

closed 2016-12-06 by loiclefort

loiclefort, 2016-12-07:
Fixed in commit 3097e278e111ba7b22df4bf30c4e1e7032f330bf