atmelcorp/atmel-software-package #25: ASP2.7 broke usleep() which broke drivers like sdmmc that use usleep

dandersonee, 2017-01-03:
Simplification of timer.c in release 2.7 was not implemented correctly and broke the drivers resulting in incorrect delays and infinite loops.

Utils/timer.c-timer_configure() has line:
rc = (tc_get_available_freq(timer->tc, tc_clks) * timer->freq) / timer->resolution;

for SAMA5D2-XULT this calculation is (83000000 * 1000) / 1000000.
So rc gets set to 0xA21=2593

Timer.c calls tc_get_cv(_sys_timer.tc, _sys_timer.channel)
This will return the value of the CV register which counts from 0-2592 then resets to 0 every 1 ms.  Each count of CV is not 1 microsecond.
The timer driver does not scale the counts in timer_usleep() to get microseconds.

Next issue:
The timer_usleep()  routine does not properly handle counter rollover for microseconds.
If the current count value  was 2400 and you asked it to wait 500usec,
It adds 2400+500=2900  and keeps reading CV until it gets to 2900 (but since CV only counts up to 2592 then rolls over to 0 it will never reach it, stuck in infinite loop.

Next issue:
In several places in utils/timer.c the following syntax is used:
  while (((int64_t)_timeout_msec - timer_get_tick()) > 0) {
I don’t believe this compiles the way you intended.
When the compiler  subtracts an uint64_t from an int64_t, it implicitly promotes the int64_t back to an uint64_t, performs the subtraction, and the result is uint64_t which you are then using for a comparison.
I believe the syntax you are looking for is to subtract the two values as uint64_t and allow rollover, then cast the result to int64_t before the comparison.
New syntax: 
  while ((int64_t)(_timeout_msec - timer_get_tick()) > 0) {

This happens in timer_sleep() and in timer_usleep() with both uint32_t and uint64_t 

Here is a report from PC-Lint on-line of a test sample:
FlexeLint for C/C++ (Unix) Vers. 9.00L, Copyright Gimpel Software 1985-2014 
--- Module: intro.txt (C)
     1  #include "stdint.h" 
     2  #include "stdio.h" 
     3  uint64_t ua;
     4  uint64_t ub;
     5  uint64_t uc;
     6  int64_t sc;
     7  int main()
     8  {
     9    ua = 0;
    10    ub = 1;
                               _
    11    sc = (int64_t)ua - ub;
intro.txt  11  Info 737:  Loss of sign in promotion from long long to unsigned long long
intro.txt  11  Info 713:  Loss of precision (assignment) (unsigned long long to long long)
                               _
    12    if (((int64_t)ua - ub) < 0)
intro.txt  12  Info 737:  Loss of sign in promotion from long long to unsigned long long
intro.txt  12  Warning 685:  Relational operator '<' always evaluates to 'false'
intro.txt  12  Warning 568:  non-negative quantity is never less than zero
    13      printf("negative\r\n");
    14    else
    15      printf("positive\r\n");
    16  
    17  }

/// Start of Pass 2 ///

--- Module: intro.txt (C)
     1  #include "stdint.h" 
     2  #include "stdio.h" 
     3  uint64_t ua;
     4  uint64_t ub;
     5  uint64_t uc;
     6  int64_t sc;
     7  int main()
     8  {
     9    ua = 0;
    10    ub = 1;
    11    sc = (int64_t)ua - ub;
    12    if (((int64_t)ua - ub) < 0)
    13      printf("negative\r\n");
    14    else
    15      printf("positive\r\n");
    16  
    17  }

--- Global Wrap-up

Warning 552:  Symbol 'sc' (line 6, file intro.txt) not accessed
Info 714:  Symbol 'uc' (line 5, file intro.txt) not referenced

dandersonee, 2017-01-03:
use of the volatile keyword in timer_sleep() and timer_usleep() for the calculated timeouts seems unnecessary.  These memory locations are not changed by interrupts or hardware unknown to the compiler.

loiclefort, 2017-01-03:
Thanks, we will fix these issues ASAP.

lfazio, 2017-01-19:
Fixed in 94cea324f98f4ec3157d233b8c0701c548d788f1.

closed 2017-01-19 by lfazio