atmelcorp/atmel-software-package #26: Not all SAMA5D2 processors have CAN

dandersonee, 2017-01-03:
The SAMA5D26 does not have the MCAN peripheral that the full SAMA5D27 has.

Suggest modifying scripts/Makefile.vars.sama5d2
Change CONFIG_HAVE_MCAN = y
TO: CONFIG_HAVE_MCAN ?= y

This allows specifying CONFIG_HAVE_MCAN = n in the project Makefile and the Makefile.vars.sama5d2 only sets CONFIG_HAVE_MCAN if it has not already been set by the project makefile.

If you are in the process of an alternative that will set it based on the processor selection, that would be even better.

Release 2.7 also added some CAN specific routines to chip.c and chip.h
I think these functions should have qualifiers based on CONFIG_HAVE_MCAN because mcan.o might not be in the build.
#if defined(CONFIG_HAVE_MCAN)
/**
 * \brief retrieve MCAN ID from its base address
 * \return MCAN ID on success, ID_PERIPH_COUNT otherwise
 */
extern uint32_t get_mcan_id_from_addr(const Mcan* addr, uint8_t int_idx);

/**
 * \brief retrieve MCAN base address from its ID
 * \return MCAN base address on success, 0 otherwise
 */
extern Mcan* get_mcan_addr_from_id(const uint32_t id);
#endif

loiclefort, 2017-01-03:
Thanks, we already have fixed this issue for the 2.8 release that we expect to ship end of next week.

lfazio, 2017-01-19:
Fixed in d316f7a1a969ca56841432626465d3ca9beae969.

closed 2017-01-19 by lfazio