atmelcorp/atmel-software-package #31: LWIP configuration

dandersonee, 2017-01-22:
Lwipopts.h is a file that is used to configure the LWIP stack.  
It is a file that should probably be copied into a project directory and customized.  
Because it is currently located in a directory with other library files that directory must be in the project include file search path.  When building in IAR, some library functions are using the copy in lib/lwip/softpack/include/arch and other files are using the copy in the project directory.
Should the library copy be moved to a different directory or do you have another suggestion?

lfazio, 2017-01-23:
Indeed, it is a configuration files related to a project and not specific to the lib.

Copying this file in another location is what we will do, but we must find a solution for both gcc and IAR environment. I had this thought when integrating the update of LWIP, but I have no good solution.

Anyway, we may keep a copy of a default configuration (maybe a better more generic than the one currently included) as of today in lwip/sofptack/include/arch. With gcc and its include path, it is pretty easy to use another lwipopts.h files located in the example, but IAR does not support files with same name in the workspace so using include path is not complete.

lfazio, 2017-01-23:
Proposal is done in commit: 6c8841a.

@dandersonee What do you think of this proposal?

dandersonee, 2017-01-23:
There are many ways it can be done, with advantages and disadvantages.
This proposal does not provide defaults, it provides a fixed built in set of options for the library.
The option must be actively selected in the Makefile (unless you plan to modify one of the Makefile scripts to set CONFIG_LIB_LWIP_DEFAULT_CONFIG="y" by default)
Opt.h already provides defaults and allows any individual option to be overridden by the project file lwipopts.h.  It does force the user to at least provide an empty lwipopts.h but that does make the user aware that it needs to be configured and how it should be configured (in that file).
If you truly want to add another layer of library defaults, maybe they should be implemented like opt.h where it only provides a define for each one if the user has not already defined it, allowing each of the library defaults to be overridden instead of an all or nothing approach.   Then leave CONFIG_LIB_LWIP_DEFAULT_CONFIG undefined so the build error tells the user to provide lwip_conf.h to configure the project layer options.  (Maybe dont need to specify CONFIG_LIB_LWIP_DEFAULT_CONFIG)

Or you could just forget about the added layer and provide the recommended lwipopts.h in the example project and maybe a lwipopts_template.h in the library directory with comments to copy to the project directory and rename it to lwipopts.h.
Since your example shows how to override the built-in set of options (as it should) even it doesn’t use the library built-in set of options.

lfazio, 2017-01-30:
The idea behind the intermediary layer is to provide a default stack configuration working for standard use without any further configuration. But this configuration can be the one in the example. Then we can move the lwipopts.h from the softpack port to examples.

loiclefort, 2017-02-03:
Commit pushed to master (dc400bee098487c165d89d8bbc5b5999c9ce15a5).
Only a single lwipopts.h file is provided now and it is located in example/eth_lwip directory.

closed 2017-02-03 by loiclefort