atmelcorp/atmel-software-package #38: usleep lockup

dandersonee, 2017-03-06:
I have seen a few cases where code gets stuck in usleep during driver initialization.
Usually when running under debugger or with caches disabled.
This happens with small values passed to usleep.
The debugger confirmed that the timer counter is still counting but usleep is waiting for TC_SR_CPCS but the counter has already passed the compare value.
What seems to be happening is that by the time usleep writes the compare value to rc, the timer counter has already advanced past the compare value and usleep sits waiting for TC_SR_CPCS that is not coming for a very long time.
I have a modification to usleep that should make it safer.  It adds COMPILER_BARRIER() to make sure compare value gets written out and reads the counter again to make sure not already past the deadline before waiting for the TC_SR_CPCS flag:

void usleep(uint32_t count)
{
	uint32_t rc;
	uint32_t status;

	if (count < 2)
		count = 2;

	/* Disable interrupts */
	arch_irq_disable();

	/* Configure RC */
	rc = ROUND_INT_DIV((_timer.channel_freq / 1000) * count, 1000);
	rc += tc_get_cv(_timer.tc, _timer.channel);
	tc_set_ra_rb_rc(_timer.tc, _timer.channel, 0, 0, &rc);
	COMPILER_BARRIER();
        // If not already past deadline wait for rc compare status
        if ((int32_t)(rc - tc_get_cv(_timer.tc, _timer.channel)) > 0) //handles rollover 
        {
          /* Wait for RC compare */
          do {
		status = tc_get_status(_timer.tc, _timer.channel);
		if (status & TC_SR_COVFS)
			_timer.upper++;
          } while ((status & TC_SR_CPCS) == 0);
        }
	/* Re-enable interrupts */
	arch_irq_enable();
}

comments/changes welcome.

lfazio, 2017-03-14:
Hi @dandersonee, after thinking to your patch, I think I will fix this by avoiding checking TC_SR_CPCS and instead doing a loop on tc_get_cv() directly.

lfazio, 2017-03-14:
Fixed in 91b972739112ef8774a4205074768f3ad825a670.

Thank you for the bug report.

closed 2017-03-14 by lfazio

dandersonee, 2017-03-14:
@lfazio see commit comments.  There is a problem with this change.