atmelcorp/atmel-software-package #41: SPI CS Question

jparsons-topcon, 2017-05-22:
I've been configuring an SPI device on the sama5d2-xplained Rev.B board, and I've noticed that there seems to be only 4 selectable CS lines per SPI device (NPCS0.. 1.. ect).
I want to run more than 4 devices on one bus and obviously this would not let me do that.
I feel I am missing a key component here as it doesn't really make sense to limit an spi bus to only 4 devices all the way up to the chip design.

Just FYI I know I can use GPIO to do more devices but I haven't found a way to incorporate that with the softpack bus object.

Basically what I'm asking is if there is something I am missing? I am mostly curious as I've never really seen it done this way.

xzp114, 2017-05-23:
Maybe you can use both SPI0 and SPI1. If so there are four devices you can drive.

jparsons-topcon, 2017-05-23:
I do understand that but that limits me to 8 total devices. Right now I'm just figuring the setup is for ease of use for applications that don't use more than 8 devices.

closed 2017-05-24 by jparsons-topcon