atmelcorp/atmel-software-package #51: PMIC ACT8945A sys0 register bitfields are defined in the reverse order

dandersonee, 2018-02-08:
Bitfields for PMIC ACT8945A register sys0 are listed in msb to lsb order when all other registers are correctly listed in lsb to msb order.
drivers/power/act8945a.c:

```
Existing Code:
// SYS @0x00
union _sys0 {
    struct {
        uint8_t
            trst:       1,  // Reset time out 0->260 1->65ms
            nsysmode:   1,  // response of SYSLEV voltage detector, 1->int 0>shutdown
            nsyslevmsk: 1,  // 1->unmask int
            nsysstat:   1,  // 1 if vsys < syslev voltage threshold
            syslev:     4;  // defines SYSLEV voltage threshold
        } bits;
    uint8_t u8;
};

Fixed Code:
// SYS @0x00
union _sys0 {
    struct {
        uint8_t
            syslev:     4,  // defines SYSLEV voltage threshold
            nsysstat:   1,  // 1 if vsys < syslev voltage threshold
            nsyslevmsk: 1,  // 1->unmask int
            nsysmode:   1,  // response of SYSLEV voltage detector, 1->int 0>shutdown
            trst:       1;  // Reset time out 0->260 1->65ms
        } bits;
    uint8_t u8;
};

```

TonyHan11, 2018-02-09:
Hi Derrick,
Thanks a lot for the fix.
We will integrate the fix in the next release.

TonyHan11, 2018-02-14:
Fixed in 7711577
Close the issue.

closed 2018-02-14 by TonyHan11