atmelcorp/atmel-software-package #52: LCDC driver calls to lcdc.c-_clear_dma_desc are called with the wrong register index

dandersonee, 2018-02-16:
The second parameter to lcdc.c-_clear_dma_desc is *dma_head_reg but all calls incorrectly passed in the LCDC_XXXXADDR register instead of the LCDC_XXXXHEAD register.

This causes these lines to write to the wrong registers:
	dma_head_reg[2] &= ~LCDC_BASECTRL_DFETCH;
	dma_head_reg[3] = (uint32_t)desc;

The calls to the companion function _set_dma_desc all properly pass in layer->reg_dma(_X)_head

The 11 places in lcdc.c that call this static internal function _clear_dma_desc should be fixed to pass in LCDC_XXXXHEAD.  Please do not patch the function to accommodate the wrong register index that was passed in.

TonyHan11, 2018-02-27:
Hi Derrick,
Thanks a lot for your report.
We have confirmed the parameters used when calling _clear_dma_desc() are wrong, they should be updated to LCDC_XXXXHEAD.
We will fix it and it will be included in the next release.

TonyHan11, 2018-05-18:
Fixed in 0b22915
Close the issue.

closed 2018-05-18 by TonyHan11