atmelcorp/atmel-software-package #53: PIO Debounce filter not configured correctly

dandersonee, 2018-03-11:
PIOA GPIO pin input filters are enabled by PIO_CFGRx.IFEN
PIOCFGRx.IFSCEN selects between glitch filter and debounce filter
If PIOCFGRx.IFSCEN=0 the glitch filter is enabled
If PIOCFGRx.IFSCEN=1 the debounce filter is enabled.
Setting the IFSCEN bit without the IFEN bit has no effect.

In the current libraries if pio_configure is called with PIO_DEBOUNCE, the  PIO_CFGRx.IFEN bit is not being set and the debounce filter is not enabled.

pio4.c-pio_configure
Current code:
```
		if (pin->attribute & PIO_DEGLITCH)
			cfgr |= PIO_CFGR_IFEN_ENABLED;

		if (pin->attribute & PIO_DEBOUNCE)
			cfgr |= PIO_CFGR_IFSCEN_ENABLED;

```
Suggested change:
```
		if (pin->attribute & PIO_DEGLITCH)
			cfgr |= PIO_CFGR_IFEN_ENABLED;

		if (pin->attribute & PIO_DEBOUNCE)
			cfgr |= (PIO_CFGR_IFEN_ENABLED  | PIO_CFGR_IFSCEN_ENABLED);

```
Some of the wording in the datasheet is unclear/confusing but I believe this is the correct interpretation based on testing SAMA5D26 with pulse generator to filter out narrow pulses.

TonyHan11, 2018-03-12:
Hi Derrick,

You are correct that PIO_CFGR_IFEN should be enabled when PIO_DEBOUNCE is needed.

We will fix it by updating code [**if (pin->attribute & PIO_DEGLITCH)**](https://github.com/atmelcorp/atmel-software-package/blob/752f1ed37c7c5cc81e4b225e50bc61b8d60be127/drivers/gpio/pio4.c#L240) to "**if (pin->attribute & (PIO_DEGLITCH | PIO_DEBOUNCE))**", and it'll align with the code in [pio3.c](https://github.com/atmelcorp/atmel-software-package/blob/752f1ed37c7c5cc81e4b225e50bc61b8d60be127/drivers/gpio/pio3.c#L288).

Thanks for your report and the patch will be applied in the next release.

dandersonee, 2018-03-12:
Thanks Tony,  that solution will work also.

TonyHan11, 2018-05-18:
Fixed in 21b38fd
Close the issue.

closed 2018-05-18 by TonyHan11