atmelcorp/atmel-software-package #54: About the SAMA5d2-Xplained emmc R/W test.

WayneVVz, 2018-04-17:
Now i'am evaluating the SAMA5D2-Xplained emmc API R/W performance by using this software-package v2.12. There are some differences between the Ex.\usb_mass_storage(UMS) and Ex.\sdcard. In  the UMS, mount a filesystem(FAT32) and use PC software ATTO Disk Benchmarks can get R:13 MB/s. But in the sdcard, reading a file with 50MB in onborad emmc by using read_file() only get R:2 MB/s.What caused this gap? Can anyone help me solve this problem? Use the IAR IDE.

Android4SAM, 2018-06-22:
We believed it related to you may using the embedded file system which based on 512 bytes sectors, which may be not that efficient.

WayneVVz, 2018-06-25:
Thanks, changing the sectors to 64k bytes could solve this problem.

closed 2018-07-04 by WayneVVz