atmelcorp/atmel-software-package #56: Analog header mismatch in drivers folder.

BlamKiwi, 2018-05-18:
`drivers/analog/acc.c` and `drivers/analog/adc.c` expect their header files to be in the `peripherals` folder. However the header files are in the same folder as the source files.  This can be easily fixed by moving the header or changing the include.

TonyHan11, 2018-05-18:
Unlike other software's architecture there's no dedicate folders for the header files, so the header files are all put in the same folders for the source files currently, thank you for your comment.

BlamKiwi, 2018-05-18:
That's not the problem. You have missing header files or bad paths. 

For example, in `acc.c` you have the following includes:

    #include "chip.h"
    #include "peripherals/acc.h" 

But there is no such `acc.h` in the `peripherals` folder. Not only that, the `Acc` type is never defined in the `acc.h` that does exist.

TonyHan11, 2018-05-18:
Yes, the header files's path really needed to be fixed, actually the `Acc` type is defined in `target\samxxxx\component\component_acc.h`.
Similar issues would be fixed in the next release, thank you very much.

TonyHan11, 2018-06-07:
Fixed in 054f25c
Close the issue.

closed 2018-06-07 by TonyHan11