atmelcorp/atmel-software-package #57: SDMMC driver issue in HS200 mode

niallleonard, 2018-05-21:
I've been using the example code to test out our own board based on SAMA5D2 Xplained. Our board has an eMMC part fitted capable of HS200 operation. However I found a bug in the sdmmc driver that prevents the code switching to that mode.
Function sdmmc_control() lines 1351-1354 look wrong:

		else if (byte == SDMMC_TIM_MMC_HS200
		    && (set->regs->SDMMC_CA0R & (SDMMC_CA0R_V18VSUP
		    | SDMMC_CA0R_V30VSUP | SDMMC_CA0R_V33VSUP))				
		    == SDMMC_CA0R_V18VSUP

I think this should be:

		else if (byte == SDMMC_TIM_MMC_HS200
			&& set->regs->SDMMC_CA0R & SDMMC_CA0R_V18VSUP

Now the device will attempt to switch to HS200 is the capabilities support it.

niallleonard, 2018-05-22:
That's great - thanks for the response.

TonyHan11, 2018-06-07:
Fixed in 4ef9b12
Close the issue.

closed 2018-06-07 by TonyHan11