atmelcorp/atmel-software-package #58: SDMMC example code

coogsie, 2018-05-23:
I am having trouble with the SDMMC example on the sama5d4-xplained board, the example fails to identify the card and subsequently cannot read/write to it. I have verified that both the board and card work by testing them using the Linux4SAM demo for these boards with the SD card being identified by Linux and showing up as a valid block device so the issue must be with the example code.

I placed printouts in various places in the example code and API and found that commands are being sent and acknowledged by the SD card and verified this using a digital logic analyser to capture some of the signals being sent to the SD card slot.

One thing I noted was that the CLK going to the SD card isn't 400KHz as stated in the code and serial terminal but instead 200KHz but I haven't looked at it much past that since I figure its not the main issue.

Is anyone able to verify the SD example code functions as expected on other boards? this would help narrow down the cause of the issue.

coogsie, 2018-05-25:
Update: I looked at the commands being sent and acknowledged by the SD card and found that everything works correctly up until ACMD51 is sent. The code always returns the no response error when the signal captured shows that the SD card is returning a valid response.

After ACMD51 returns the no response error the code then attempts to send CMD55 again which causes the SAMA5D4 to freeze, the CMD55 is not sent as it does not show up on the CMD signal trace however there is a packet on the DAT0 line. The attached picture shows this, the packet on the CMD trace is ACMD51 with the response from the SD card and the packet on the DAT0 line is the mystery packet.

[image]

andyjefflin, 2018-05-28:
1. About the 400KHz issue, you are right, we need use pmc_get_peripheral_clock() instead of pmc_get_master_clock() to get peripheral clock in drivers/sdmmc/hsmcid.c. 
The following modification will be applied in the next release to fix this issue.
```
static void hsmci_set_device_clock(struct _hsmci_set *set, uint32_t freq)
{
	assert(set);
	assert(freq);

	Hsmci *regs = set->regs;
	uint32_t div, new_freq;
-	uint32_t mck = pmc_get_master_clock();
+	uint32_t pck = pmc_get_peripheral_clock(set->id);

-	if ((mck % freq) == 0)
-		div = mck / freq;
+	if ((pck % freq) == 0)
+		div = pck / freq;
	else
-		div = (mck + freq) / freq;
-	new_freq = mck / div;
+		div = (pck + freq) / freq;
+	new_freq = pck / div;

	/* Modify MR */
	hsmci_set_clock(regs, div, 0x7);
	set->dev_freq = new_freq;
}
```
2.  read/write failed in your side
I am not very clear about the  ACMD51/CMD55 causes the SAMA5D4 to freeze, could you tell me how the processor stuck?
Anyway, I am sure the sdmmc examples can work on the sama5d4-xplained board with MCI1 interface.

coogsie, 2018-05-30:
I have had a look at this issue some more, reading through the SD card specifications I realised that the ACMD51 response is sent over the DAT0 line which explains what the packet in the scope trace is. With this in mind I quickly commented out the code relevant to the ACMD51 call and found that other commands which expect data back from the SD card are also broken.

Looking through the code and placing printf calls in various parts I believe I have isolated the cause of the problem to code which sets up the DMA. From what I can tell the DMA controller never seems to think that data is recieved causing the _SendCmd call to sit in the 30 second timeout loop and eventually timeout since the flags it checks are never set.

andyjefflin, 2018-05-31:
We need more information to help you on problem:
1.Which release softpack do you use? Please redownload the latest release softpack for test.
2.Do you have modify the code, before you do the test?
3.Which type of SD card do you use?
4.Please send the all meassage display in the pc terminal to me?

coogsie, 2018-05-31:
> 1.Which release softpack did you use? Please redownload the latest release softpack for test.

I am using the latest softpack
>2.Did you have modified the code before you did the test?

No
>3.Which type of SD card do you use?

Sandisk 16GB SDHC
>4.Please send all the meassage display in the pc terminal to me.

`-I- Device clk 400kHz`
`-E- Cmd52 ERR_NO_RESPONSE` 
`-E- Cmd5 ERR_NO_RESPONSE`
`-I- Device supports 3.0V:[XXXXXX] 3.3V:[XXXXXX]`
`-I- SDHC MEM`
`-E- Cmd51 ERR_NO_RESPONSE`
`-E- Acmd51 ERR_NO_RESPONSE`
After the last message the code becomes unresponsive.

The cause of the code becoming unresponsive is the mutex lock check at line 193 of hsmcid.c
`	while (mutex_is_locked(&set->dma_unlocks_mutex)) {`
`		if (set->use_polling)`
`			dma_poll();`
`	}`
This mutex is supposed to be unlocked by the hsmci dma callback function (line 327 of hsmcid.c) triggered by the XDMAC interrupt, however since the interrupt never occurs the mutex is never unlocked. This causes the code to sit in the while loop indefinitely.

To summarize
1 - ACMD51 is sent to SD card
2 - SD card sends command response on CMD line and then the data on the DAT0 line
3 - data is never detected by the HSMCI or the XDMAC isn't configured correctly
4 - (3) results in the hsmci dma callback never being called since the XDMAC interrupt never occurs
5 - (4) means that mutex isn't unlocked and other flags aren't cleared
5 - code sits in 30 second timeout (line 393 of sdmmc_api.c) waiting for the SD card to send the data packet since flags are never cleared
6 - code attempts to send next command, falls into loop at line 193 of hsmcid.c and never gets out since mutex wasn't unlocked and never will be

andyjefflin, 2018-06-01:
From your information:
1.	CMD8 and ACMD41 command and response are ok. It only uses CMD pin to send command and receive response.
2.	ACMD51 failed and the mutex always locked:
ACMD51 try to read SCR information from SD card by DAT0 pin, it is failed to read data.
The mutex locked because the HSMCI read/write FIFO overflow. But the WRPROOF and RDPROOF in HSMCI_MR can stop the HSMCI Clock during read/write access if the fifo is full. So, I think it is not the code problem. 
I advise you to double check the hardware connection. You can change a new SD card or use another SAMA5D4-CPLAINED board for test.

andyjefflin, 2018-06-07:
1. the 400KHz issue fixed in  56a812e777eab043a5af06964e517399bdb169be
2. Read/Write failed problem didn't appear in my side, please check the hardware connection on your board.
Close the issue.

closed 2018-06-07 by TonyHan11