atmelcorp/atmel-software-package #59: Enabling PIO interrupt should not read `PIO_ISR`...

alfred-ai, 2018-06-07:
...because it will clear other pending PIO interrupt on the same port. This issue was founded on one of the cortex-M7 chips in the SAME/S/V family.

https://github.com/atmelcorp/atmel-software-package/blob/1e7c0f735fd33a1e88fbfc452310c6aef116ed5a/drivers/gpio/pio3.c#L491

https://github.com/atmelcorp/atmel-software-package/blob/1e7c0f735fd33a1e88fbfc452310c6aef116ed5a/drivers/gpio/pio4.c#L405

https://github.com/atmelcorp/atmel-software-package-samv7/blob/31a206b0ab290128ef05d790dde5278ea4edad94/samv7/libraries/libchip/source/pio_it.c#L311

TonyHan11, 2018-06-14:
Hi Alfred,
Thanks for your report.
Reading PIO_ISR would clear the status of pending PIO interrupt, so the two "pio->PIO_ISR;" should be removed from pio_enable_it(), the fix would be integrated soon.

TonyHan11, 2018-06-21:
Fixed in commit 74a5d89, close the issue.

closed 2018-06-21 by TonyHan11