atmelcorp/atmel-software-package #60: Overflow and rounding errors with ADC trigger period

iguffick, 2018-07-03:
I'm new to GitHub, so apologies if I'm reporting this in the wrong way!

I've found a problem with the library code for setting the trigger period of the ADC.
The function 'adc_set_trigger_period' in drivers/analog/adc.c does not seem to work as intended.

After debugging, this is due to overflows in the calc for the ADC_TRG period value.
The current code uses:
	trg_period = period * (_adc_clock/1000) - 1;

With my current configuration '_adc_clock' is 5857142, so any value of 'period' over 11 will overflow the 16 bit 'trg_period' value that gets written to ADC_TRG period.

I'd suggest a change to:
	uint64_t trg_period;
	trg_period = (((uint64_t)period * (uint64_t)_adc_clock) / (uint64_t)1000000) - 1;

This solves the overflow and removes some rounding errors. The resultant call to 'adc_set_trigger_period' is then in units of micro-seconds.
So the function call 'adc_set_trigger_period(1000);' will set a 1000us (1ms) period that results in an ADC trigger rate of 1000Hz. With my suggested code this now works.

TonyHan11, 2018-07-05:
Hi Ian,

Thank you for your report.

Your method of calculation could improve the accuracy of ADC trigger period, and it would be integrated into the master branch later(patch attached bellow).

For the overflow issue, it's limited by ADC_TRGR.TRGPER, using ADC clock with lower frequency should solve the problem.

```
diff --git a/drivers/analog/adc.c b/drivers/analog/adc.c
index 358004e..fd40de0 100644
--- a/drivers/analog/adc.c
+++ b/drivers/analog/adc.c
@@ -548,10 +548,11 @@ void adc_set_tracking_time(uint32_t dwNs)
 
 void adc_set_trigger_period(uint32_t period)
 {
-	uint32_t trg_period;
+	uint64_t trg_period;
+
 	if (_adc_clock == 0)
 		return;
-	trg_period = period * (_adc_clock/1000) - 1;
+	trg_period = (((uint64_t)period * (uint64_t)_adc_clock) / 1000000) - 1;
 	ADC->ADC_TRGR = (ADC->ADC_TRGR & ~ADC_TRGR_TRGPER_Msk) | ADC_TRGR_TRGPER(trg_period);
 }
 
diff --git a/drivers/analog/adc.h b/drivers/analog/adc.h
index 31a41dc..6df9ef1 100644
--- a/drivers/analog/adc.h
+++ b/drivers/analog/adc.h
@@ -339,7 +339,7 @@ extern void adc_set_tracking_time(uint32_t dwNs);
 
 /**
  * Sets the trigger period.
- * \param period Trigger period in nS.
+ * \param period Trigger period in uS.
  */
 extern void adc_set_trigger_period(uint32_t period);
 
diff --git a/drivers/analog/adcd.c b/drivers/analog/adcd.c
index 75caab9..161a2ea 100644
--- a/drivers/analog/adcd.c
+++ b/drivers/analog/adcd.c
@@ -191,7 +191,7 @@ static void adcd_configure(struct _adcd_desc* desc)
 		case TRIGGER_MODE_ADC_TIMER:
 			/* Trigger on internal ADC timer */
 			adc_set_trigger_mode(ADC_TRGR_TRGMOD_PERIOD_TRIG);
-			adc_set_trigger_period(250);
+			adc_set_trigger_period(250000);
 			break;
 		default :
 			break;
```

dandersonee, 2018-07-05:
By implementing it in this way, have you not broken backwards compatibility by changing the API?
When someone pulls the newer version of the library to get bug fixes, this would break existing code
that calls adc_set_trigger_period and passes in a value in units of milliseconds (comment in adc.h incorrectly said ns - probably copy/paste error from function above it or typo).
They would not even get a compile error making it obvious the API changed and they need to fix something in their code.
Would it not be better to add a new function adc_set_trigger_period_us(uint32_t period) and leave the existing function for backwards compatibility?  Existing function could change the calculation internally but should not change the units of the parameter.

TonyHan11, 2018-07-06:
Hi Derrick,
Thanks for your comment. 
Yes, it breaks the compatibility of the API, and I'll rename adc_set_trigger_period() to adc_set_trigger_period_ns() so the compilation errors would remind users to update the parameters accordingly.

dandersonee, 2018-07-09:
There is a minimum and maximum ratio that must be maintained between the adc_clock and the trigger_period to fit in the 16-bit field that the hardware uses.  An assert should probably be added in adc_set_trigger_period to check that these limits are not violated.  The fact that the demo example violated these limits does not mean that customers using the library did not pass valid arguments.  It does not change that the current checkin breaks backwards compatibility (and even worse does so silently) for customers using the library.  I just think there are better ways of implementing it without breaking backwards compatibility and have made the arguments for why it is important to consider them.

Currently the demo in examples/adc/main.c sets adc_clock to 300000.
The adcd.c uses a default value of 250 ms for the trigger period if the user configures the adc using adcd_configure() (which this demo does).  
These values appear to fall outside the 16-bit ratio. (250 ms would be 75000 counts of a 300000 Hz clock and 75000=0x124F8 which doesn’t fit in 16-bits. 
I think I have the calculations correct.
Suggest reducing adc_clock to 100000 in examples/adc/main.c and adding the asserts to check the ratio in the library.

I still consider that my original arguments and suggestions remain valid.

iguffick, 2018-07-10:
Hi Derek,
In principal I agree with not breaking something that already exists (and works).
But I think it's so broken as to be not fit for purpose, in that case how much should you try to retain?

In order to work out how it should be done I made a spreadsheet to calc the TRGPER for a number of ADC CLOCK values and wanted periods. Using the existing calcs, the actual values were you get the correct result are very particular. If using a high clock speed, the multiplication result quickly overflows the 16 bit value. Equally if using a low clock speed, you'd be setting a much slower period so the period value is higher and also quickly overflows. Using the new method give correct calculation, they are not unattainable trigger periods.
The comment that it is passed a time in ns is also clearly wrong, it seems to be in ms. This is also part of the problem, you couldn't set a 20MHz ADC clock and ask for a 100ms period.

By leaving the old function in place, it looks like a valid use-able and working function.
By all means rename the new function so that compilation fails, but the old function if retained should be marked with a #warning 'this doesn't work, please use xxxxx function'.

dandersonee, 2018-07-10:
This triggering method is not a universal solution that fits all situations.  It has a limited dynamic range that must be considered.  That is why I think it is important to add the assert statements to the library function that will catch the programmer trying to use it in an inappropriate situation but they compile out when you turn off debug mode for production.  The adc example shows how the other various triggering methods (using software, external timer, etc) can be configured for other situations where the clock and the triggering period are too far apart.  This method works for slow sampling with a moderate clock or fast sampling with a fast clock.  If you really need slow sampling and a fast clock then one of the other methods is more appropriate.  For using this method the parameters of adc_clock and trig_period should be thoughtfully chosen, not just random.  The necessity of using 64 bit multiplication and division while more accurate is more time consuming and may be arguable, I haven’t worked through the cycle times.  With carefully chosen parameters they are probably overkill for most situations.  I did not object to using the 64-bit calculations, only the keeping the function name the same while changing the units of a parameter.
Using the functions at the adc.c level works fine.  The API for using this trigger method at the adcd.c level where it has a fixed 250ms setting is not very flexible.

TonyHan11, 2018-08-14:
Many API modifications of softpack would break the compatibility in some degree, when users update softpack to another version the git history and change-log would be very helpful. During the developing period there are many chances to find the trigger period mismatching with the expected one and using proper clock is very important to reach the correctness and accuracy. A assert statement for the value written to ADC_TRGR is really useful for avoiding the overflow case.

closed 2018-08-14 by TonyHan11