atmelcorp/atmel-software-package #61: SDMMC in HS200 mode slower then HSDDR mode?

WayneVVz, 2018-07-04:
I saw a commit is to fix the conditions checked for MMC's HS200 mode support. I following this commit change my code, emmc dose running in HS200 mode but slower then HSDDR mode. What happened? Core is SAMA5D27-D1G. When running HSDDR mode the R is 80M/s.
Our project is using the package v2.12, but with some change.

WayneVVz, 2018-07-04:
here is the log
 MSG 1407: Using a 1-bit data bus
 MSG 1475: Clocking the device at 400000 Hz
 MSG 1475: Clocking the device at 34666666 Hz
 MSG 1407: Using a 8-bit data bus
 MSG 1407: Using a 8-bit data bus
 MSG 1475: Clocking the device at 104000000 Hz
 MSG 1407: Using a 1-bit data bus
 MSG 1475: Clocking the device at 400000 Hz
 MSG 1475: Clocking the device at 34666666 Hz
 MSG 1407: Using a 8-bit data bus
 MSG 1407: Using a 8-bit data bus
 MSG 1475: Clocking the device at 104000000 Hz
 MSG 1407: Using a 1-bit data bus
 MSG 1475: Clocking the device at 400000 Hz
 MSG 1475: Clocking the device at 34666666 Hz
 MSG 1407: Using a 8-bit data bus
 MSG 1407: Using a 8-bit data bus
 MSG 1475: Clocking the device at 104000000 Hz

junminlin, 2018-07-06:
The more important factor for SD/MMC's IO performance is at the memory devices side. To be more specically, in case of reading, the time waiting for the response start bit of the data streaming. If your project need dig more detailed, please contact the support hotline with the wave form you see on your board.

WayneVVz, 2018-07-10:
Ok, the problem is solved. But I found another problem. Running in HS200 mode with clock 208MKz, a small chance of MmcInit in f_mount will get stuck here. Change the mode to HSDDR, there is no such situation.

Android4SAM, 2018-07-10:
D2 datasheet (DS60001476B-page 1759) said that:
Support for e.MMC HS200 (Maximum SDCLK Frequency = **120 MHz**).
We agree that we need to fail gracefully. But, we don't have eMMC support HS200 in hand. Is it possible to share the part number you are testing?

WayneVVz, 2018-07-10:
Thanks, for your reminder.
Although frequency is set to one of 104MHz, 120MHz, 208MHz. This problem has also appeared. The program seems to be stuck on _HwSetClock(). The e.MMC we use is KLM4G1WEPD. I will run a test on softpack v2.13 tomorrow. By the way, when the frequency is 104MHz, the speed is less than freq = 120MHz. HS200 base on 104 MHz is much slower than HSDDR.

BA8819, 2018-07-14:
When running in HS200 mode, eMMC chips require 1.8V VCCQ. You need to make sure that the software is handling the PMIC LDO output properly to 1.8V because it won't work on any values above e.g. 3.3V.

nirvann, 2018-10-01:
Also, please mind that e.MMC devices do **not** support switching to low signaling voltage on the fly. This implies, in case the host dynamically changes VCCQ voltage from 3.3V to 1.8V, that the e.MMC device is then fully powered down (both VCC and VCCQ), and powered up again.
So, in order to use HS200 bus speed mode, the e.MMC device requires 1.8V VCCQ at power-on time. Is this condition fulfilled?

If yes: when initialization freezes do you get the `"%u tuning blocks."` debug trace?

What is the frequency of the on-board oscillator that clocks the SAMA5D27-D1G?

Samsung's KLM4G1WEPD e.MMC is given to comply with the e.MMC v5.0 Standard, up to HS400 bus speed mode, with sequential read speed reaching 100 MB/sec.
The HS200 mode should increase the interface bandwidth to 120 MB/sec (compared to 104 MB/sec in HS DDR mode at 52 MHz).
Are there errors during transfer?

closed 2019-03-26 by WayneVVz