atmelcorp/atmel-software-package #62: Random ERR_IO on eMMC tests

BA8819, 2018-07-14:
Hi

I'm doing a memory test on the eMMC part and occasionally I'm having the read or write function returning an error (ERR_IO). 
There is no details on what ERR_IO actually means.

WayneVVz, 2018-08-01:
You should run the eMMC demo from softpack v2.13 to make sure your code is not wrong. If it does't work, check your hardware IO connection.

andyjefflin, 2018-08-09:
Hi:
Use CMD19/CMD14 command to detect the data pins of emmc/sd like below two steps:
In a first step, the host sends CMD19 to the Device, followed by a specific data pattern on each selected data lines. As a second step, the host sends CMD14 to request the Device to send back the reversed data pattern. With the data pattern sent by the host and with the reversed pattern sent back by the Device, the functional pins on the bus can be detected.
So the processor check in turn from 8, 4, 2, to 1 data pins until the correct function pins detected. if the eMMC connector support 4 data pins not 8 pins, it will return CMD19 ERR_IO first time.
Maybe that is why you get error (ERR_IO). Did you have the same situation? If yes, it is not an issue.

closed 2018-09-13 by TonyHan11

nirvann, 2018-09-26:
Calculating the maximum block program time of this particular e·MMC leads to a value of 1.6 s (assuming the e·MMC is clocked at 50 MHz).
This is the block write timeout value, and currently it has to be hardcoded in the `sdmmc_initialize` function. While the default value (500 ms) is fine with SDXC cards, when interfacing with an e·MMC we should verify whether 500 ms is long enough.
That's why, here, we increase the timeout value from 500 ms to 2 s.