atmelcorp/atmel-software-package #66: invalid CAN rate in SAMA5D2-Xplained example

majbthrd, 2018-10-05:
The example in ./examples/can/ can be built for SAMA5D2-Xplained with:

make TARGET=sama5d2-xplained

That build configuration invokes ./targets/sama5d2/board_sama5d2-xplained.h, which has:

#define BOARD_CAN_BUS0_FREQ 1000000
#define BOARD_CAN_BUS1_FREQ 1000000

However, when this image is programmed into a SAMA5D2-Xplained (both Rev B and Rev C tried), it actually generates a data rate closer to 877kHz (14% error).

This problem might be hidden to the developer if only cross-connecting two boards with the same wrong configuration, but is plainly visible when checking the waveform with a scope or trying to interface to something else.

andyjefflin, 2018-10-09:
Hi:
Please modify the value of quanta in mcand_set_baudrate() like below:
const struct _mcan_quanta quanta = {
		.before_sp = 1 + 2 + 1,
		.after_sp  = 1,
		.sync_jump = 4,
	};

majbthrd, 2018-10-10:
I can confirm a correct bit rate with the finer quanta of { 4, 1, 4 } versus the { 16, 5, 5 } hard-coded into the as-is driver.

I also had the same correct bit rate operation using { 15, 12, 4 }, values that I arrived at independently

Is there any particular design rule of thumb to pick certain values over another (above and beyond providing the correct bit rate timing given a particular bus speed)?  The documentation is a bit thin on this topic.

Is there any particular reason why { 16, 5, 5 } was hard-coded into the MCAN driver?  I ask because an earlier revision of this reference code allowed the user layer to set these values via the mcan_initialize() call (and there was no mcan_bitrate() function).  Or, was internalizing these values just a side effect of abstracting away the API whether MCAN or CAN peripherals were used?

Thanks.

andyjefflin, 2018-10-16:
This issue can be fixed to select UPLLCK for CAN clock.
It is recommended to use the CAN clock at frequencies of 20, 40 or 80 MHz. To achieve these frequencies, PMC GCLK3 must select the UPLLCK (480 MHz) as source clock and divide by 24, 12, or 6. GCLK3 allows the system bus and processor clock to be modified without affecting the bit rate communication. 
So, the code modified in drivers/can/mcand.c  as below:
```
const struct _mcan_quanta quanta = {
-	.before_sp = 1 + 2 + 12,
-	.after_sp  = 12,
+	.before_sp = 2 + 25,
+	.after_sp  = 1 + 12,
	.sync_jump = 4,
};
const struct _mcan_quanta quanta_fd = {
-	.before_sp = 1 + 2 + 5,
-	.after_sp  = 5,
+	.before_sp = 2 + 12,
+	.after_sp  = 1 + 5,
	.sync_jump = 2,
};

struct _pmc_periph_cfg cfg = {
	.gck = {
-		.css = PMC_PCR_GCKCSS_MCK_CLK,
-		.div = 1,
+		.css = PMC_PCR_GCKCSS_UPLL_CLK,
+		.div = 6,
	},
};
```

majbthrd, 2018-10-16:
UPLLCK has the obvious disadvantage of depending on the clocks being configured for USB, which is an unnecessary power drain (and hopefully the reason why this approach is not the de facto one employed in the code).

Regardless, I must assume that it is somehow not practical for Atmel to check in a fix to the example code to work at the data rate it purports to operate at.  Hopefully, someone in the future who is confounded by this particular problem will somehow find this issue.

closed 2018-10-16 by majbthrd

andyjefflin, 2018-10-22:
Fixed in 286a1f7899df767257e4f9cbab80fbd546397254