atmelcorp/atmel-software-package #68: SDMMC CMD17 fails due to ACMD error

dmgstz, 2018-12-10:
My system has a eMMC, with a FAT filesystem on top with a 512 cluster size. I'm encountering the error below ocasionally when trying to open files (f_open in FatFs):

```
-W- CMD17 ended with error flags 0100, cmd status ERR_NO_RESPONSE
-E- Cmd17 ERR_NO_RESPONSE
-E- Cmd17(0x40) ERR_NO_RESPONSE
```

ERR_NO_RESPONSE translates to Auto CMD Timeout according to the flags the code checks. The eMMC supports block count so I assume that the command fails due to a previous Auto CMD 23 rather than Auto CMD 12, I am still diagnosing why this might occur and I wanted to raise the issue here as it seems weird to me that the log mentions that CMD17 fails with Auto CMD error, looks like it is not synchronized to me as CMD17 is a Single Block Read operation which should not need of any CMD23 or CMD12?

I am using polling mode in the library, let me know your thoughts.

Thanks.

nirvann, 2018-12-10:
Hi. As you've found out already, the Auto CMD error probably occurs further to the preceding command, either SET_BLOCK_COUNT+READ_MULTIPLE_BLOCK or SET_BLOCK_COUNT+WRITE_MULTIPLE_BLOCK.
Would you enable the debug traces, in order to confirm which was the previous command?
Any case, one hypothesis at this point is the DAT line timeout, possibly being too short w.r.t. the e.MMC device. Would you have a look at Issue #62 and let us know whether the patch discussed there is applied on your version?

dmgstz, 2020-11-11:
I think it is fine to close this one, as I have not been able to reproduce in more than two years :)