atmelcorp/atmel-software-package #69: ATSAMA5D225C-D1M DDR2 support

friesendrywall, 2018-12-22:
I don't see any support code for the internal DDR2 on the ATSAMA5D225C-D1M part with 16MB ddr2.  Could this be added?

TonyHan11, 2018-12-24:
Hi friesendrywall,
I didn't catch on the internal DDR2 on the ATSAMA5D225C-D1M, as I know there's no internal DDR2 on it (https://www.microchipdirect.com/product/search/all/ATSAMA5D225C-D1M). Would you like to explain it in detail? Thanks.
User could fond the DDR supported at https://github.com/atmelcorp/atmel-software-package/blob/v2.14/scripts/Makefile.vars.sama5d2#L102, and support for new DDR devices could be added after it has been validated.

friesendrywall, 2018-12-24:
See here https://www.microchip.com/wwwproducts/en/ATSAMA5D225C-D1M and download the datasheet, it has a 128mbit ddr2 in the top list.

TonyHan11, 2018-12-24:
According to chapter "Reference Documents" in the datasheet (http://ww1.microchip.com/downloads/en/DeviceDoc/SAMA5D2-Plus-DDR2-LPDDR2-System-in-Package-(SIP)-60001484b.pdf) the "2 Mwords × 4 Banks × 16 bits DDR2 SDRAM (128 Mbit)" is W9712G6KB, the support to it hasn't been added to softpack yet, it'll be added after validated on the according SIP. You could keep watching at https://github.com/atmelcorp/atmel-software-package/blob/master/ChangeLog.md for the updates, thank you.

friesendrywall, 2018-12-24:
What do you mean, validated? This seems to be in production and would be useless without ddr.

Since my design is complete with PCB in hand, could I get any suggestions as to how to add this myself?

friesendrywall, 2018-12-24:
Or would it be better to open a microchip support ticket?

mattwood2000, 2018-12-24:
The SIP devices are validated from a silicon perspective as they are in production.  From a software perspective, the eval boards and "generic" targets are supported by the software package.  Since there is no D1M eval board you will have to add the support yourself.  I would start by:

1.  Adding the W9712G6KB timing parameters and memory type definition to drivers/extram/ddram.{c,h} 
2.  Create a new board to match your hardware in target/sama5d2 (adapt the SOM1-EK or generic target).
3.  Modify or create a new ddram linker script in target/sama5d2/toolchain/gnu.  If you look at ddram.ld, you can see it is generically setup to support 64Mbytes + 16MBytes of DDR which you will not have in the D1M device.
4.  Modify the Makefile rules to include your new board type, and a few other things I probably forgot :)

You can open a ticket if you like, but I suspect Microchip adds support on a priority basis and since there are no eval boards for the D1M, having them do it will take time.  You will be better off adding the support yourself.

TonyHan11, 2018-12-25:
The D1M SIP has already been validated before production, but it will take time for the softpack team to add them one by one into the softpack, and the support for D1M would be added immediately after softpack team got the timing parameters.
For how to add the support mattwood2000 has described it very clearly bellow, a lot of thanks to mattwood2000:)

TonyHan11, 2019-06-25:
The update for supporting DDR2 W9712G6KB (16MB) on the ATSAMA5D225C-D1M are in the these commits: fd14df8 420fe90 353ab92

closed 2019-06-25 by TonyHan11