atmelcorp/atmel-software-package #73: EMMC5.1 init failed

WayneVVz, 2019-03-26:
Hello,
  Due to material updates, our boards are mixed with two versions of eMMC, version 5.0 and version 5.1 . In ver 5.0 , every thing is normal , But in ver 5.1, eMMC init failed with error ERR_NO_RESPONSE.
  Please help me to fix this case. Soc is SAMA5D27D5M. 

  [image]

  Thanks!!

BA8819, 2019-04-10:
Did you make sure that your mmc card is compatible with 5.0 and ealier versions? the driver doesn't support >V4.0, it will try to find if the card can run at version 4 speeds/modes.

nirvann, 2019-06-26:
I don't see any obvious reason why a v5.1 e.MMC wouldn't be backward-compatible.
Have you sought support from the e.MMC Manufacturer?