atmelcorp/atmel-software-package #74: Sample clock tune freeze (MMC card running HS200 with speed up 120MHz)

BA8819, 2019-04-10:
Hi

I'm trying to run Panasonic mmc card (8Gb) with ATSAMA5D28C. Initially everything looks okay but on recent tests when rebooting the board multiple times, the card initialization freezes; I looked into it and it seems that the card is freezing when trying to set the clock speed (HS200) above 95Mhz. The failure is related to the sample_tune_procedure which has to be done in HS200 when running high speed. 
The issue isn't consistent as the initialisation will work sometimes. I looked in the Errata and this issue is acknowledged however couldn't seem to see any implementation in the latest driver. 

I tried adding the condition hc2r & SDMMC_HC2R_EXTUN  to the while loop in the sdmmc_tune_sampling but that didn't get me anywhere and in fact the failure rate is worse now.

when forcing the driver to work with lower speed modes (DDR or SDR) there are no problems observed.

BA8819, 2019-04-10:
Think I got it fixed by adding this condition (properly!) to the sdmmc_tune_sampling 
[image]

nirvann, 2019-04-10:
A software workaround to the Erratum hasn't been released yet.
Basically, the workaround consists in modifying the while() loop so it stops if the Controller clears the _Check Execute Tuning_ flag:
`while (hc2r & SDMMC_HC2R_EXTUN && !(regs->SDMMC_NISTR & SDMMC_NISTR_BRDRDY)) ;`

However some additional code may be required to ensure that (1) the last _SEND_TUNING_BLOCK_ command isn't in progress anymore, and (2) the _Buffer Read Ready_ event, when it occurs, is consumed.