atmelcorp/atmel-software-package #87: freertos不支持MMU，而examples移植的freertos却启用了MMU。应该不对。

swhan1610, 2020-04-08:
WEAK void board_init(void)
{
#ifdef VARIANT_DDRAM
	bool ddram = false;
#else
	bool ddram = true;
#endif

#ifdef VARIANT_SRAM
	bool clocks = true;
#else
	bool clocks = false;
#endif

	/* Configure misc low-level stuff */
	board_cfg_lowlevel(clocks, ddram, true);  // 启用MMU

board_cfg_lowlevel(clocks, ddram, true);这句应改为：
board_cfg_lowlevel(clocks, ddram, false);
但是，这样改了，IAR下载程序后却跑不起来了。
不知是何原因？请指教!(VARIANT_DDRAM我也定义了）

swhan1610, 2020-04-08:
1, I want an freertos example of running in DDRAM without MMU.  on the board sama5d2-xplained. 
2, Does the USB function must need the MMU?
thx!

TonyHan11, 2020-04-08:
hi,
In softpack there are no address translations between physical address and virtual address, they are just the same so MMU could be used with freertos without issues.

swhan1610, 2020-04-09:
> 
> 
> hi,
> In softpack identical address translation is used between physical address and virtual address, the addresses are just the same so MMU could be used with freertos without issues.

Why MMU must be enabled？for USB ？

TonyHan11, 2020-04-10:
Some USB descriptor's length is odd, this cause some unaligned access problem when cache is disabled. Therefore, if we put --no_unaligned_access to some USB code (for example *.c) to avoid unaligned access, we can disable MMU. But, disable MMU cost a lot of performance drop, therefore, we suggest use MMU enabled.

swhan1610, 2020-04-14:
> 
> 
> Some USB descriptor's length is odd, this cause some unaligned access problem when cache is disabled. Therefore, if we put --no_unaligned_access to some USB code (for example lib/usb/common/usb_descriptors.c) to avoid unaligned access, we can disable MMU. But, disable MMU cost a lot of performance drop, therefore, we suggest use MMU enabled.

atmel-software-package，Do you have any related forums for this library(atmel-software-package)? In this way, we can ask each other questions. 
thx.

TonyHan11, 2020-04-16:
For any issues about software package you can ask for support using support hotline or https://microchip.my.salesforce.com/, and creating tickets in [github](https://github.com/atmelcorp/atmel-software-package/issues/new) is OK too, thanks.

closed 2020-04-20 by TonyHan11