atmelcorp/atmel-software-package #91: chip_pins.h wrong SSC0 group B

mhanuel26, 2020-05-23:
Hi,

Just notice this under sama5d2

`/* ========== Pio PIN definition for SSC0 peripheral ========== */

#define SSC0_IOS1_TX (PIO_PB20C_TK0 | PIO_PB21C_TF0 | PIO_PB22C_TD0)
#define SSC0_IOS1_RX (PIO_PB23C_RD0 | PIO_PB24C_RK0 | PIO_PB25C_RF0)
#define PINS_SSC0_IOS1 {\
	{ PIO_GROUP_C, SSC0_IOS1_TX | SSC0_IOS1_RX, PIO_PERIPH_C, PIO_DEFAULT },\
}`

and I think it should be PIO_GROUP_B as my best understanding.

TonyHan11, 2020-05-25:
Thanks for your report. The typos may fixed by the following patch:
```
--- a/target/sama5d2/chip_pins.h
+++ b/target/sama5d2/chip_pins.h
@@ -734,10 +734,10 @@
 	{ PIO_GROUP_C, SSC0_IOS1_TX | SSC0_IOS1_RX, PIO_PERIPH_C, PIO_DEFAULT },\
 }
 
-#define SSC0_IOS2_TX (PIO_PC12E_TK0 | PIO_PC13E_TF0 | PIO_PB2PIO_PC14E_TD02C_TD0)
+#define SSC0_IOS2_TX (PIO_PC12E_TK0 | PIO_PC13E_TF0 | PIO_PC14E_TD0)
 #define SSC0_IOS2_RX (PIO_PC15E_RD0 | PIO_PC16E_RK0 | PIO_PC17E_RF0)
 #define PINS_SSC0_IOS2 {\
-	{ PIO_GROUP_C, SSC0_IOS2_TX | SSC0_IOS2_RX, PIO_PERIPH_E, PIO_DEFAULT },\
+	{ PIO_GROUP_E, SSC0_IOS2_TX | SSC0_IOS2_RX, PIO_PERIPH_E, PIO_DEFAULT },\
 }
 
 /* ========== Pio PIN definition for SSC1 peripheral ========== */
@@ -745,13 +745,13 @@
 #define SSC1_IOS1_TX (PIO_PA14B_TK1 | PIO_PA15B_TF1 | PIO_PA16B_TD1)
 #define SSC1_IOS1_RX (PIO_PA17B_RD1 | PIO_PA18B_RK1 | PIO_PA19B_RF1)
 #define PINS_SSC1_IOS1 {\
-	{ PIO_GROUP_A, SSC1_IOS1_TX | SSC1_IOS1_RX, PIO_PERIPH_B, PIO_DEFAULT },\
+	{ PIO_GROUP_B, SSC1_IOS1_TX | SSC1_IOS1_RX, PIO_PERIPH_B, PIO_DEFAULT },\
 }
 
 #define SSC1_IOS2_TX (PIO_PB14C_TK1 | PIO_PB15C_TF1 | PIO_PB16C_TD1)
 #define SSC1_IOS2_RX (PIO_PB17C_RD1 | PIO_PB18C_RK1 | PIO_PB19C_RF1)
 #define PINS_SSC1_IOS2 {\
-	{ PIO_GROUP_B, SSC1_IOS2_TX | SSC1_IOS2_RX, PIO_PERIPH_C, PIO_DEFAULT },\
+	{ PIO_GROUP_C, SSC1_IOS2_TX | SSC1_IOS2_RX, PIO_PERIPH_C, PIO_DEFAULT },\
 }
 
 /* ========== Pio PIN definition for TC0 peripheral ========== */
```

TonyHan11, 2020-05-26:
Fixed by commit [75c997a](https://github.com/atmelcorp/atmel-software-package/commit/75c997ab508e1d5615a04587ab6cd9d8a9eb337d)

closed 2020-05-26 by TonyHan11