atmelcorp/atmel-software-package #92: Can someone include the device support for AT91SAM9260/AT91SAM9260?

gowthamkalangi, 2020-06-12:
I have been searching for the software package of AT91SAM9260 / AT91SAM9260. 
As of now, the software package does support SAM9X60.
Can someone include the device drivers for AT91SAM9260 / AT91SAM9260 as they are pin to pin compatible, so can be used with minor changes if needed.

TonyHan11, 2020-06-12:
You could ask for 9G20 / 9260 support on https://microchipsupport.force.com/

LeSpocky, 2024-08-06:
> I have been searching for the software package of AT91SAM9G20 / AT91SAM9260. As of now, the software package does support SAM9X60. Can someone include the device drivers for AT91SAM9G20 / AT91SAM9260 as they are pin to pin compatible, so can be used with minor changes if needed.

You should check the datasheets again. SAM9G20 and SAM9X60 both have an Arm926EJ-S Processor, but packages and pinout differ quite significantly.

I'd like to see sam9g20 support as well, but I'm afraid you have to use sam-ba v2.18 or some other version 2 release for that.