atmelcorp/atmel-software-package #93: SAMAD3XPLAINED nand flash issue

gurpreetshanky, 2020-06-14:
Hello,

i'm trying to erase the flash in this board using sam-ba tool. I removed the NAND CS jumper and connected the USB to laptop and pressed reset button, upon which I can see monitor mode by prseeing V# on serial port. When I run this command

sam-ba -p serial -b sama5d3-xplained -a nandflash -c erase

I get following error

Opening serial port 'COM6'
Connection opened.
C:/Dev/embedded_rust/sam-ba_3.3.1-win32/sam-ba_3.3.1/qml/SAMBA/Applet.qml:247: Error: Could not initialize applet (status: undefined)
Connection closed.

Any ideas or Am I doing something wrong?

TonyHan11, 2020-06-15:
Have you ever tried to using sam-ba with the jumper(JP5) connected?
According to page 15 of [the user guide](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11269-32-bit-Cortex-A5-Microcontroller-SAMA5D3-Xplained_User-Guide.pdf), when the jumper is removed the /CE will be high, and the NAND flash is disabled, the applet could not detect the NAND.

gurpreetshanky, 2020-06-15:
@TonyHan11 okk I made changes to my steps

1. connected usb to serial port of DEBUG port
2. power supply to board using a direct adapter 
3. remove the nand cs jumper and pressed reset
4. check on the serial port if its in monitoring mode(it is)
5. connect the nand jumper back
6. run the program

still its the same error

TonyHan11, 2020-06-15:
Here are some requests needed before erase NAND with sam-ba from serial port:
 - connect to PC/laptop with USB through J6 (USB-A)
 - power the board with a 5V adapter through J2 (DC 5V_IN)
 - keep JP5 (NAND CS) connected
 - an additional serial port occurs after the USB is connected, then SAM-BA commands (such as "V#") could be executed through this port

Now ```sam-ba -p serial -b sama5d3-xplained -a nandflash -c erase``` should work properly. Before execute the command please keep the serial port closed, otherwise the following error would occur:
"Error: Could not open serial port 'COMx': Access is denied."

Following is the log of erase NAND:
```
Opening serial port 'COMx'                            
Connection opened.                                     
Detected memory size is 268435456 bytes.               
Page size is 2048 bytes.                               
Buffer is 22528 bytes (11 pages) at address 0x0030a4a0.
NAND header value is 0xc0902405.                       
Supported erase block sizes: 128KB                     
Executing command 'erase'                              
Erased 131072 bytes at address 0x00000000 (0.05%)      
Erased 131072 bytes at address 0x00020000 (0.10%)      
Erased 131072 bytes at address 0x00040000 (0.15%)      
... ...
Erased 131072 bytes at address 0x0ffc0000 (99.95%)
Erased 131072 bytes at address 0x0ffe0000 (100.00%)
Connection closed.
```

gurpreetshanky, 2020-06-15:
Hey @TonyHan11 I was able to flash it. I didn't even have to use the j2 port , just the j6 port. thanks.

closed 2020-06-15 by gurpreetshanky