atmelcorp/atmel-software-package #94: Flashing application to SAM9x60-ek

zalasimas, 2020-06-23:
For a while learning to work with SAM9x60-ek board. Maybe someone could tell me how can I upload compiled program to the board? I try to upload usart example program from _atmel-software-package/examples/usart_. After compiling the example, I found a folder called _build_ in the example directory. How can I upload program to the board? And which file should I upload? I tried to upload _usart.bin_ file into the board using SAM-BA, following this command _sam-ba -p serial:COM21 -b sam9x60-ek -a nandflash -c erase::0x40000 -c writeboot:usart.bin_, but nothing happened. After restarting the board roomBOOT message is shown.

TonyHan11, 2020-06-24:
First of all, occording to [datasheet ch 11.4.6.4 Arm Exception Vectors Check](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM9X60-Data-Sheet-DS60001579B.pdf) the binary generated by software package is not suitable for using as a boot due to the missing of binary size at the sixth vector (offset 0x14), the value is set to 0 in [the source code](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sam9x60/toolchain/gnu/cstartup.S#L75)(for IAR it's the same). The binary from [at91bootstrap](https://github.com/linux4sam/at91bootstrap) would be a good choice.
Before booting from NAND a correct ```Boot Configuration Packet``` should be configured, otherwise an internal default Boot Configuration Packet will be used. For sam9x60-ek you could configure it by the following commands (make sure the OTP works properly):
```
sam-ba -p serial:COM21 -b sam9x60-ek -a bootconfig -c writecfg:bscr:EMULATION_ENABLED
sam-ba -p serial:COM21 -b sam9x60-ek -a bootconfig -c writecfg:bcp-emul:DBGU,NFC_IOSET1
``` 
And there might be [a booting issue (Chapter 5.2)](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM9X60-EK-UG-DS50002907B.pdf) related with NAND on SAM9X60-EK board.

zalasimas, 2020-06-25:
Thanks for the answer Tony. 

When this command are executed
```
sam-ba -p serial:COM21 -b sam9x60-ek -a bootconfig -c writecfg:bcp-emul:DBGU,NFC_IOSET1
```
I am receiving this message: 

_C:\Users\MB\Desktop\sam-ba_3.3.1>sam-ba -p serial:COM21 -b sam9x60-ek -a bootconfig -c writecfg:bcp-emul:DBGU,NFC_IOSET1                                                                                    Opening serial port 'COM21'                                                                                                                                                                                        Connection opened.
Buffer is 23647 bytes (Infinity pages) at address 0x003073a0.
Executing command 'writecfg:bcp-emul:DBGU,NFC_IOSET1'
Setting BCP-EMUL to DBGU,NFC_IOSET1
C:/Users/MB ERIDAS/Desktop/sam-ba_3.3.1/qml/SAMBA/Applet/BootConfigApplet.qml:70: Error: Write Boot Config command failed (status=15)
Connection closed._     

First command executed successfully.

closed 2020-09-03 by TonyHan11