atmelcorp/atmel-software-package #95: Error in SECUMOD definitions

markstz, 2020-06-24:
Some of the bit definitions in target\[target]\component\secumod.h are incorrect. This applies to the sama5d2 and sama5d4 targets.

```
#define SECUMOD_CR_AUTOBKP_Pos 5
#define SECUMOD_CR_AUTOBKP_Msk (0x3u << SECUMOD_CR_AUTOBKP_Pos) /**< \brief (SECUMOD_CR) Automatic Normal to Backup Mode Switching */
#define SECUMOD_CR_AUTOBKP(value) ((SECUMOD_CR_AUTOBKP_Msk & ((value) << SECUMOD_CR_AUTOBKP_Pos)))
#define SECUMOD_CR_AUTOBKP_SW_SWITCH   (0x2u << 3) /**< \brief (SECUMOD_CR) When in Normal mode, software must switch to Backup mode before powering down the core */
#define SECUMOD_CR_AUTOBKP_AUTO_SWITCH (0x1u << 3) /**< \brief (SECUMOD_CR) When in Normal mode, the power down of the core supply will automatically switch the mode to Backup mode, simultaneously with core to backup isolation barrier activation. (default) */
```
The SECUMOD_CR_AUTOBKP_SW_SWITCH and SECUMOD_CR_AUTOBKP_AUTO_SWITCH definitions are only bit-shifting 3 places, instead of 5 or SECUMOD_CR_AUTOBKP_Pos places.

```
#define SECUMOD_CR_SCRAMB_Pos 9
#define SECUMOD_CR_SCRAMB_Msk (0x3u << SECUMOD_CR_SCRAMB_Pos) /**< \brief (SECUMOD_CR) Memory Scrambling Enable */
#define SECUMOD_CR_SCRAMB(value) ((SECUMOD_CR_SCRAMB_Msk & ((value) << SECUMOD_CR_SCRAMB_Pos)))
#define SECUMOD_CR_SCRAMB_DISABLE (0x2u << 3) /**< \brief (SECUMOD_CR) Memories are not scrambled */
#define SECUMOD_CR_SCRAMB_ENABLE  (0x1u << 3) /**< \brief (SECUMOD_CR) Memories are scrambled (default) */
```
Similar thing for SECUMOD_CR_SCRAMB_DISABLE and SECUMOD_CR_SCRAMB_ENABLE, they should be shifting 9 places.

TonyHan11, 2020-06-28:
Thank you for the issue you reported. 
According to the datasheet you are correct, would you like to create a pull request for the fix? Thank you in advanced.

TonyHan11, 2020-07-01:
Fixed in [f370be5fa](https://github.com/atmelcorp/atmel-software-package/commit/f370be5fa5fd2f98333e1ac8e1d991567a0425d3), thanks.

closed 2020-07-01 by TonyHan11