atmelcorp/atmel-software-package #96: Issues with sdmmc-sdcard example on SAMA5D3-Xplained

PaulMadle, 2020-09-07:
Hi, 

I am trying to bring up the SD Card example project. 

I have got the code running and have turned the TRACE level up to DEBUG... I can see the following.... when I type "i" (Display device info) using a serial terminal program:

-D- ../../drivers/sdmmc/hsmcid.c:542 SDMMC_IOCTL_POWER(1)
-D- ../../drivers/sdmmc/hsmcid.c:109 Power the device on
-D- ../../drivers/sdmmc/hsmcid.c:542 SDMMC_IOCTL_SET_HSMODE(0)
-D- ../../drivers/sdmmc/hsmcid.c:622 Using timing mode 0x00
-D- ../../drivers/sdmmc/hsmcid.c:542 SDMMC_IOCTL_SET_BUSMODE(1)
-D- ../../drivers/sdmmc/hsmcid.c:594 Using a 1-bit data bus
-D- ../../drivers/sdmmc/hsmcid.c:542 SDMMC_IOCTL_SET_CLOCK(400000)
-D- ../../drivers/sdmmc/hsmcid.c:631 Clocking the device at 400000 Hz
-I- Device clk 400 kHz
-D- ../../lib/libsdmmc/sdmmc_api.c:597 PwrON
-D- ../../lib/libsdmmc/sdmmc_api.c:377 Cmd0(0)
-D- ../../drivers/sdmmc/hsmcid.c:719 cmd 0, op 0x0001, 0 x 0, arg 0x00000000, status 0

The code then hangs at this point.... 

I've also stuck some breakpoints in the interrupt routine in the hsmcid.c code... The code is not getting there. It seems that the code needs the ISR to clear the drivers state variable.

Any help is much appreciated :)

TonyHan11, 2020-09-08:
I've run the "sdmmc_sdcard" example on sama5d3-xplained board and could not re-produce the issue you described, no breakpoints are used during my test, only modified the TRACE_LEVEL as the following:
```
diff --git a/drivers/sdmmc/hsmcid.c b/drivers/sdmmc/hsmcid.c
index 9157bb65..b100c9f3 100644
--- a/drivers/sdmmc/hsmcid.c
+++ b/drivers/sdmmc/hsmcid.c
@@ -39,6 +39,6 @@
 # define TRACE_LEVEL SDMMC_DRV_TRACE_LEVEL
 #elif TRACE_LEVEL > 4
 # undef TRACE_LEVEL
-# define TRACE_LEVEL 4
+# define TRACE_LEVEL 5
 #endif

diff --git a/lib/libsdmmc/sdmmc_api.c b/lib/libsdmmc/sdmmc_api.c
index 5c38d0af..bbc6b74b 100644
--- a/lib/libsdmmc/sdmmc_api.c
+++ b/lib/libsdmmc/sdmmc_api.c
@@ -40,6 +40,6 @@
 # define TRACE_LEVEL SDMMC_LIB_TRACE_LEVEL
 #elif TRACE_LEVEL > 4
 # undef TRACE_LEVEL
-# define TRACE_LEVEL 4
+# define TRACE_LEVEL 5
 #endif
```
Following is the log:
```
-- SD/MMC Example --
Softpack v2.17 (v2.17-dirty)
Built for sama5d3-xplained
Processor: SAMA5D36
Processor clock: 528 MHz
Master clock: 132 MHz
MMU is enabled
I-Cache is enabled
D-Cache is enabled

SD Card menu:
   h: Display this menu
   t: Toggle between Slot[0](SD/MMC) and Slot 1 (microSD)
   i: Display device info
   l: Mount FAT file system and list files
   r: Read the file named 'test_data.bin'
   w: Perform a basic RAW read/write test.

-D- ../../drivers/sdmmc/hsmcid.c:543 SDMMC_IOCTL_POWER(1)
-D- ../../drivers/sdmmc/hsmcid.c:109 Power the device on
-D- ../../drivers/sdmmc/hsmcid.c:543 SDMMC_IOCTL_SET_HSMODE(0)
-D- ../../drivers/sdmmc/hsmcid.c:622 Using timing mode 0x00
-D- ../../drivers/sdmmc/hsmcid.c:543 SDMMC_IOCTL_SET_BUSMODE(1)
-D- ../../drivers/sdmmc/hsmcid.c:594 Using a 1-bit data bus
-D- ../../drivers/sdmmc/hsmcid.c:543 SDMMC_IOCTL_SET_CLOCK(400000)
-D- ../../drivers/sdmmc/hsmcid.c:631 Clocking the device at 400000 Hz
-I- Device clk 400 kHz
-D- ../../lib/libsdmmc/sdmmc_api.c:597 PwrON
-D- ../../lib/libsdmmc/sdmmc_api.c:377 Cmd0(0)
-D- ../../drivers/sdmmc/hsmcid.c:721 cmd 0, op 0x0001, 0 x 0, arg 0x00000000, status 0
-D- ../../drivers/sdmmc/hsmcid.c:233 XFRDONE
-D- ../../lib/libsdmmc/sdmmc_api.c:377 Cmd52(80000c08)
-D- ../../drivers/sdmmc/hsmcid.c:721 cmd 52, op 0x0252, 0 x 0, arg 0x80000c08, status 0
-D- ../../drivers/sdmmc/hsmcid.c:208 HSMCI_SR 0x04100024
-E- Cmd52 ERR_NO_RESPONSE
-D- ../../lib/libsdmmc/sdmmc_api.c:377 Cmd0(0)
-D- ../../drivers/sdmmc/hsmcid.c:721 cmd 0, op 0x0002, 0 x 0, arg 0x00000000, status 0
-D- ../../drivers/sdmmc/hsmcid.c:233 XFRDONE
-D- ../../lib/libsdmmc/sdmmc_api.c:377 Cmd8(1aa)
-D- ../../drivers/sdmmc/hsmcid.c:721 cmd 8, op 0x0172, 0 x 0, arg 0x000001aa, status 0
-D- ../../drivers/sdmmc/hsmcid.c:208 HSMCI_SR 0x04100024
-E- Cmd8 ERR_NO_RESPONSE
[... 226 lines cut ...]
 .RD_BL_MALIGN               0
 .DSR_IMP                    0
 .C_SIZE                     0xF59
 .RD_CUR_MIN                 0x7
 .RD_CUR_MAX                 0x6
 .WR_CUR_MIN                 0x7
 .WR_CUR_MAX                 0x6
 .C_SIZE_MULT                0x7
 .ER_BL_EN                   1
 .SECT_SIZE                  0x7F
 .WP_GRP_SIZE                0x0
 .WP_GRP_EN                  0
 .R2W_FACT                   0x5
 .WR_BL_LEN                  0x9
 .WR_BL_PART                 0
 .FILE_FMT_GRP               0
 .COPY                       0
 .PERM_WP                    0
 .TMP_WP                     0
 .FILE_FMT                   0x0
 .CRC                        0x16

-D- ../../drivers/sdmmc/hsmcid.c:543 SDMMC_IOCTL_RESET(0)
-D- ../../drivers/sdmmc/hsmcid.c:117 Release and power the device off
```

PaulMadle, 2020-09-08:
Hi Tony.... 

Many thanks for the response... Great to know that people are maintaining this code base and helping users... Very much appreciated.

Comparing the log data, my code goes to la-la land after the first command sent to the SD card, your log prints this:

-D- ../../drivers/sdmmc/hsmcid.c:233 XFRDONE

Mine does not... This code is in the ISR for the HSMCI... So I am concluding that my problem relates to interrupts. 

I have setup everything through Eclipse on a Windows PC with the following chain to build, load, run.

- Build within eclipse using the GNU compiler.
- Load using SAM-BA into external NAND flash.
- Reset board (I have adjusted bootcmd in u-boot to target bare-metal applications).
- U-Boot copies code from NAND flash to DDR memory
- Code runs from DDR
- Connect GDB through J-Link.

I suspect that somewhere along this list of actions, I am loosing interrupt vector tables or something?

Any wisdom is much appreciated.

Many thanks,

Paul.

TonyHan11, 2020-09-08:
Maybe you can try to test the example alone without U-Boot, by launching the GDB server and running command "make TARGET=sama5d3-xplained debug" in path "examples/sdmmc_sdcard" ...

```
SEGGER J-Link GDB Server V6.12i Command Line Version

JLinkARM.dll V6.12i (DLL compiled Feb  3 2017 15:24:41)

-----GDB Server start settings-----
...
...
...
Listening on TCP/IP port 2331
Connecting to target...
J-Link found 1 JTAG device, Total IRLen = 4
JTAG ID: 0x4BA00477 (Cortex-A5)
Connected to target
Waiting for GDB connection...
```

```
softpack/examples/sdmmc_sdcard$ make TARGET=sama5d3-xplained debug
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
...
...
...
Writing register (#40 = 0x00000000)
Loading section .fixed0, size 0x1321c lma 0x20000000
Loading section .ARM.exidx, size 0x8 lma 0x2001321c
Loading section .relocate, size 0x3c lma 0x20013224
Start address 0x20000000, load size 78432
Transfer rate: 66 KB/sec, 11204 bytes/write.
Writing register (PC = 0x20000000)
(gdb)c
Continuing.
```

PaulMadle, 2020-09-08:
I have tried this out without success :( GDB/J-link appear to die prior to running the application.

I can see that the makefile is attempting to load some code into the SRAM (I am guessing the bootstrap code?) and then some other code into the DDR (the sdmmc_sdcard application).

At the moment, my SAM-BA method is putting ONLY the application into the DDR and then vectoring to 0x20000000... I am wondering if I need to also load the bootstrap into SRAM and vector to the SRAM instead? Does the bootstrap code include the interrupt vector table? - I am having trouble finding the interrupt vector table?... I am not sure how the interrupt vectors are allocated?

I am currently trying to understand what is in the bootstrap image via the makefile.

Again, wisdom is much appreciated... Thanks for the help so far ;)

PaulMadle, 2020-09-08:
Ok, so looking at it... I think I need to load the bootstrap code into SRAM and vector to the DDRAM... Otherwise the interrupts will not run through the code in cstartup.s.... I am guessing that might be why interrupts are causing the code to vector into no mans land.

Sound about right?

TonyHan11, 2020-09-09:
The [bootstrap](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sama5d3/toolchain/gnu/common.gdb#L73) ([code here](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/common/bootstrap.c#L47)) is an applet that initializes the DDRAM, after DDRAM is initialized the application could be loaded. 
No needs to load vectors to SRAM before run the application from 0x20000000, the vector tables are compiled into [DDRAM](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sama5d3/toolchain/gnu/ddram.ld#L124) and be [copied to SRAM 0x300000](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sama5d3/toolchain/gnu/cstartup.S#L207) after stack pointers are initialized, also [SRAM is remapped to 0](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sama5d3/toolchain/gnu/cstartup.S#L226) as the vector base address is 0.

PaulMadle, 2020-09-09:
Again, many thanks Tony... 

So as I understand it...  I should not need to load the bootstrap into the SRAM... The vectors are copied from the DDR... and indeed I've confirmed that this is working, I can see the vectors in DDR and then the copy being made to SRAM where they end up at address 0x0.

However, I can see that none of the ISRs (DMA, HSMCI, timers) are triggering... this has lead me to discover how the interrupts are setup and enabled. I can see that the various drivers are making calls to register there interrupts via [a registration function](https://github.com/atmelcorp/atmel-software-package/blob/master/drivers/irq/irq.c#L167) and then [enabling](https://github.com/atmelcorp/atmel-software-package/blob/master/drivers/irq/irq.c#L208) them. That all seems good. I am imagining that when an interrupt occurs, the code should vector to the [default irq handler](https://github.com/atmelcorp/atmel-software-package/blob/master/drivers/irq/irq.c#L102). However, I can see through JLink that this is not happening. 
In my build, the _default_irq_handler() has an address lifted from my map file as:

.text._default_irq_handler
                **0x200073ac**       0x8c ./build/sama5d3-xplained/ddram/drivers/drivers.a(irq.o)

I was therefore thinking that I should see this address in the list of interrupt vectors at the beginning of SRAM... However, after all the initialisation etc.... The SRAM content looks like:

[image]

Something is not right... I can see the addresses of some interrupt routines.. but not all. Once again, your wisdom is gold.

NOTE: as part of this investigation, I have switched off the compilers optimiser ( -O0 ).

TonyHan11, 2020-09-10:
According to your description it seems that the IRQ interrupt is disabled before running the example, would you like to update ["cpsie   a"](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/target/sama5d3/toolchain/gnu/cstartup.S#L249) in cstartup.S to "cpsie   ai" or "cpsie   aif" and try again? Thanks.

PaulMadle, 2020-09-10:
Hi Tony... 

Once again, many thanks for the suggestion, I gave it a go... sadly, no joy. 

I took a look at the ARM core registers to identify if the IRQ and/or fast IRQ interrupt were masked... They are both active. 
Also, I can see that at the point that I expect interrupts to occur (following the call to send SD card commands)... The code disappears somewhere... This is leading me to think that U-Boot is leaving the processor in a state that is incompatible with the example code. 

There are a few places the interrupt vector table can move, you can switch from 0x00000000 to 0xFFFF0000 somehow (through pin settings possibly - not sure) or by setting the VBAR register (Vector Base Address Register), it looks like U-Boot has code that does exactly this (see [here](https://github.com/linux4sam/u-boot-at91/blob/master/arch/arm/cpu/armv7/start.S#L75)).

So I am going to see if I can change the VBAR setting so that interrupts vector to 0x00000000 and not wherever U-Boot has placed them... Me and assembler... What could possibly go wrong!

I am working on a project that is attempting to Radiation test a few SAMA5D3s to see how they perform in a space environment... The boards will be placed in a radiation chamber. I can only have a small number of wires travelling to the board so I can't really have a J-Link JTAG interface to the board.. Hence the U-Boot setup.

TonyHan11, 2020-09-10:
There might be some unknown settings done by U-Boot, to make things easy might you could try to use [at91bootstrap](https://github.com/linux4sam/at91bootstrap) to load and run the example, it will load the binary from NAND (**address 0x40000**) to DDRAM and run.
The config for loading U-Boot could be used for loading the softpack's example, here are the commands to build at91bootstrap:
```
make mrproper
make sama5d3_xplainednf_uboot_defconfig
make CROSS_COMPILE=arm-none-eabi-
```

PaulMadle, 2020-09-10:
Aha.... I think I've fixed it! woop woop! ;) ;)

I added a few lines to cstartup.s in order to ensure that the processor is vectoring to the correct addresses (starting 0x00000000). 

These lines are after the code that copies the vector tables to SRAM from DDR....

```
/* Relocate */
	ldr     r0, =_etext
	ldr     r1, =_srelocate
	ldr     r2, =_erelocate
1:
	cmp     r1, r2
	ldrcc   r3, [r0], #4
	strcc   r3, [r1], #4
	bcc     1b

/* PRM added in an attempt to support U-Boot startup of this code */
/* ensure that the interrupt vectors are mapped to the start of the address space (0x00000000) */
	mov 	r0, #0
        mcr 	p15, 0, r0, c12, c0, 0	@Set VBAR to 0x00000000
    
/* END PRM addition */
```

U-Boot must be moving the vector tables base address (hence interrupts causing the code to vector to no-mans land).

I've essentially told the core to vector back from the default location (from 0x00000000).

Any chance that this could be incorporated (probably with a conditional compile... similar to U-Boot source) in order to make the code more resilient to processor state at startup? 

Now I can see the SD card example working nicely!

PaulMadle, 2020-09-12:
Ok, I've just started working with the FreeRTOS examples too.... same problem and same fix...

This time, the example code builds a FreeRTOS equivalent cstartup.s (cstartup_freertos.S), I placed the same fix in the equivalent location and the example is now running... Flashing LEDs etc.

Many thanks for your help on this Tony, do you think incorporating these lines might be a good idea?

TonyHan11, 2020-09-14:
Yes, It's the root cause that "Vector Base Address" is not 0 causes the issue, might "Vector Base Address" been changed somewhere in the U-Boot. Incorporating these lines ensures that the example always runs properly. Maybe you can create a pull request for it, or if you don’t mind, I can submit the code.
Many thanks for your findings and contributions:)

PaulMadle, 2020-09-14:
Many thanks for your support also... Extremely helpful.

Yes, I am very happy to perform a pull request for review. I will modify the startup.s files where applicable. This may involve the introduction of some conditional compilation but I will see how that pans out... It looks like there are many cstartup.s files, 1 for each board? Therefore, probably no need for conditional compilation... Just need to identify which chips support the relocation of the vector base address and add the assembler to the applicable cstartup.s files.

Many thanks again TonyHan11.

Does someone need to create a branch for me?

Paul.

TonyHan11, 2020-09-15:
In fact each chip has 2 cstartup files for each chip, one for GNU and another for IAR, adding the code to set VBAR to 0 to the cstartup files in target/sama5dx/toolchain is OK for all the SAMA5 chips.
You can create a branch yourself and [create a pull](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) request to the master branch.

TonyHan11, 2020-09-15:
Since all the SAMA5 chips need to set VBAR to 0x00000000, it might be better to add the code to [drivers/irq/irq.c](https://github.com/atmelcorp/atmel-software-package/blob/v2.17/drivers/irq/irq.c#L134) as follows:
```
index 6de5ce3f..a906cf74
--- a/drivers/irq/irq.c
+++ b/drivers/irq/irq.c
@@ -131,6 +131,12 @@ static void _default_irq_handler(void)

void irq_initialize(void)
{
+#ifdef CONFIG_ARCH_ARMV7A
+       uint32_t vbar = 0;
+       /* Set VBAR to 0x00000000 */
+       asm("mcr p15, 0, %0, c12, c0, 0" :: "r"(vbar));
+#endif
+
        _initialize_handlers_pool();

#if defined(CONFIG_HAVE_AIC2) || defined(CONFIG_HAVE_AIC5)
```

PaulMadle, 2020-09-15:
Ok Tony, that solution looks a bit simpler to implement... I will have a go at a pull request today! :)

TonyHan11, 2020-09-21:
solved by setting VBAR to 0 in #99

closed 2020-09-21 by TonyHan11