atmelcorp/atmel-software-package #101: get_usart_id_from_addr return error for sam9x60 flexcom12

zizimumu, 2020-10-27:
we have tested flexcom12 usart in sam9x60, found chip_common.c/get_usart_id_from_addr return error, the functions in chip_common.c does not support flexcom10/flexcom11/flexcom12 for sam9x60

TonyHan11, 2020-10-27:
Thank you very much for your report. If you can create a pull request that includes the patch, it would be greatly appreciated, otherwise we will resolve this issue later.

zizimumu, 2020-10-28:
it's not urgent, hope you can fix it at next release version.

TonyHan11, 2020-10-28:
Thank you and the patch has been integrated to the master branch by commit [75d48e8](https://github.com/atmelcorp/atmel-software-package/commit/75d48e853faf4b5f7b0afb280cd20e01543f5921)

closed 2020-10-28 by zizimumu