atmelcorp/atmel-software-package #102: sam9x60 CAN baud rate timing parameters

zizimumu, 2020-10-29:
I tested CAN example with baud rate 800K, I got the timing parameters from cand.c/cand_set_baudrate :
brp = 14, propag = 3 , phase1= 4, phase2= 5 , sjw = 3; that means  actually CAN baud rate is 833K, it's a big deviation.
but I think, "brp = 24, propag = 2 , phase1= 3, phase2= 1" will be a better timing parameters, so i think, to make more precise baudrate, cand_set_baudrate function need to be improved.

TonyHan11, 2020-10-30:
Yes, the baud rate of 800Kbps can be obtained more accurately by using the timing parameters you listed. Compared with calculation by code, it is easier for customers to manually calculate the parameters to obtain a more accurate baud rate. (Please note that the timing delay is fixed in the code and needs to be adjusted for actual usage)
If you have a better solution, please don't hesitate to share your ideas or create PRs, thank you in advance.

zizimumu, 2020-11-02:
actually, the timing parameters of 800K is from liux kernel, maybe you can refer to linux kernel drivers\net\can\dev.c\can_calc_bittiming, this function is more accurate to caculater the timing parameters.

closed 2020-11-02 by zizimumu