atmelcorp/atmel-software-package #104: Cross compiling for SAMA5D36 gives some undefined reference error

jarus0, 2020-11-12:
Changes that I made in below file
`atmel-software-package/scripts/Makefile.vars`

```
CROSS_COMPILE ?= arm-none-eabi-
to 
CROSS_COMPILE ?= arm-linux-gnueabihf-

```
and 

```
LDFLAGS = --specs=nano.specs -Wl,--cref -Wl,--check-sections -Wl,--gc-sections
to 
LDFLAGS = -Wl,--cref -Wl,--check-sections -Wl,--gc-sections
```

For compiling any given example using below cmd
`make TARGET=sama5d3-ek`

I am getting below errors
```
./build/sama5d3-ek/sram/target/sama5d3/toolchain/gnu/cstartup.o: In function `entry':
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../target/sama5d3/toolchain/gnu/cstartup.S:246: undefined reference to `__libc_init_array'
./build/sama5d3-ek/sram/examples/usb_cdc_serial/main.o: In function `_usb_data_received':
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:362: undefined reference to `iprintf'
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:368: undefined reference to `iprintf'
./build/sama5d3-ek/sram/examples/usb_cdc_serial/main.o: In function `_debug_help':
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:379: undefined reference to `iprintf'
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:380: undefined reference to `iprintf'
./build/sama5d3-ek/sram/examples/usb_cdc_serial/main.o: In function `main':
/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:514: undefined reference to `iprintf'
./build/sama5d3-ek/sram/examples/usb_cdc_serial/main.o:/home/dnk025/atmel-software-package/examples/usb_cdc_serial/../../examples/usb_cdc_serial/main.c:420: more undefined references to `iprintf' follow
/opt/gcc-linaro-arm-linux-gnueabihf-4.7-2013.04-20130415_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.7.3/../../../../arm-linux-gnueabihf/bin/ld: error: required section '.rel.plt' not found in the linker script
/opt/gcc-linaro-arm-linux-gnueabihf-4.7-2013.04-20130415_linux/bin/../lib/gcc/arm-linux-gnueabihf/4.7.3/../../../../arm-linux-gnueabihf/bin/ld: final link failed: Invalid operation
collect2: error: ld returned 1 exit status
make: *** [build/sama5d3-ek/sram/usb_cdc_serial.elf] Error 1

```

TonyHan11, 2020-11-13:
First of all, it is suggested to use the toolchain referenced in [README.md](https://github.com/atmelcorp/atmel-software-package/blob/master/README.md).
The problem encountered by arm-linux-gnueabihf is caused by the different c libraries used in the toolchains.
The following is a list of modifications (somthing may be missing) to use software package with arm-linux-gnueabihf toolchain:
- remove "bl __libc_init_array" from toolchain/gnu/cstartup.S
- remove "--specs=nano.specs" from "LDFLAGS"
- remove "-Dprintf=iprintf" from "CFLAGS"
- rewrite the "printf" routine and adjust the stack size

```
FYI:
    Partly tested OK (but NOT RECOMMENDED) with examples/getting_started on sama5d2-xplained board.
arm-linux-gnueabihf-gcc -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/tool/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: /tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/src/gcc/configure --target=arm-linux-gnueabihf --prefix= --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/build-arm-linux-gnueabihf/install//arm-linux-gnueabihf/libc --with-bugurl=https://bugs.linaro.org/ --enable-gnu-indirect-function --enable-shared --disable-libssp --disable-libmudflap --enable-checking=release --enable-languages=c,c++,fortran --with-gmp=/tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/build-arm-linux-gnueabihf/host-tools --with-mpfr=/tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/build-arm-linux-gnueabihf/host-tools --with-mpc=/tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/build-arm-linux-gnueabihf/host-tools --with-isl=/tmp/dgboter/bbs/rhev-vm8--rhe6x86_64/buildbot/rhe6x86_64--arm-linux-gnueabihf/build/build-arm-linux-gnueabihf/host-tools --with-arch=armv7-a --with-fpu=neon --with-float=hard --with-arch=armv7-a --with-pkgversion='GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)'
Thread model: posix
gcc version 8.3.0 (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36))
```

jarus0, 2020-11-18:
As you suggested commented in below file
`atmel-software-package/target/sama5d3/toolchain/gnu/cstartup.S`
> bl      __libc_init_array

now compile successfully but binary file size become 600MB for getting-started example.

TonyHan11, 2020-11-23:
The tool is NOT a suggested tool and currently there are no plans to add support for it. Would you like to point out the necessity of using this tool chain? Thanks.

jarus0, 2020-11-28:
I didn't understand, In softpack.md files written "SAMA5D3 family" supports.

In Makefile.vars files I tried below compilers.
```
CROSS_COMPILE ?= arm-none-eabi-			// Compiling but giving error 1
CROSS_COMPILE ?= arm-linux-gnueabi-		// Compiling failed giving error 2
CROSS_COMPILE ?= arm-linux-gnueabihf-		// Compiling failed giving error 3
```
error 1
Running in my sama5d3 board gives below error
`./usb_cdc_serial.bin: line 1: syntax error: unexpected "("`

error 2
`arm-linux-gnueabi-gcc: error: nano.specs: No such file or directory`

so as you guided changed `LDFLAGS`
`LDFLAGS = -Wl,--cref -Wl,--check-sections -Wl,--gc-sections`
Below error given

```
LINK build/sama5d3-xplained/sram/usb_cdc_serial.elf
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: error: build/sama5d3-xplained/sram/usb_cdc_serial.elf uses VFP register arguments, /usr/lib/gcc-cross/arm-linux-gnueabi/7/crtbegin.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc-cross/arm-linux-gnueabi/7/crtbegin.o
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: error: build/sama5d3-xplained/sram/usb_cdc_serial.elf uses VFP register arguments, /usr/lib/gcc-cross/arm-linux-gnueabi/7/libgcc.a(_udivmoddi4.o) does not
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc-cross/arm-linux-gnueabi/7/libgcc.a(_udivmoddi4.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: error: build/sama5d3-xplained/sram/usb_cdc_serial.elf uses VFP register arguments, /usr/lib/gcc-cross/arm-linux-gnueabi/7/crtend.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/7/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc-cross/arm-linux-gnueabi/7/crtend.o
./build/sama5d3-xplained/sram/target/sama5d3/toolchain/gnu/cstartup.o: In function `entry':
/home/dnk025/Music/Project_PLC/atmelSoftwarePakage/atmel-software-package/examples/usb_cdc_serial/../../target/sama5d3/toolchain/gnu/cstartup.S:246: undefined reference to `__libc_init_array'
collect2: error: ld returned 1 exit status
../../scripts/Makefile.rules:102: recipe for target 'build/sama5d3-xplained/sram/usb_cdc_serial.elf' failed
make: *** [build/sama5d3-xplained/sram/usb_cdc_serial.elf] Error 1
```

also tried with removing `__libc_init_array` from startup file in cstartup.S
didn't help

error 3 
same as error 2

TonyHan11, 2020-11-30:
For error 1:
I can't reproduce the problem and don't know how you run the binary on the sama5d3 board, would you like to show me the steps? Usually we use GDB to run the example(command ```make TARGET=sama5d3-ek debug```, see ```Run and Debug (with GDB)``` in [README.md](https://github.com/atmelcorp/atmel-software-package/blob/master/README.md)) or put the binary to the boot media and boot the board from the media.