atmelcorp/atmel-software-package #105: problem in exicuting .bin file in SAMA5d3 borad

pragnesh036, 2020-12-03:
I am using SAMA5D3 customized board with buildroot os 
my  test-pio example compile successfully 
but when I execute my bin file on my bord I will get an error like 

```
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
# ./test-pio.bin
./test-pio.bin: line 1: ��: ./test-pio.bin: line 1: can't open 0�: no such file
./test-pio.bin: line 1: ���������������������: not found
./test-pio.bin: line 1: ��: not found
./test-pio.bin: line 1: v0: not found
./test-pio.bin: line 1: ��Q�0�40�4���:l��l��: not found
��M�@-�0�/�@��Ѝ�P����!���0�����N�@-��O�@-�D������������!�P-�
                            ���8���@���o���������0��S��/0��S��/�/��v0�v0$��$��0A����C������/0��S��/�/��v0�v0@-�,@��0��S�����0��S���: not found
not found
./test-pio.bin: line 1: ��lP��: not found
./test-pio.bin: line 2: @-������: not found
./test-pio.bin: line 1: ��@P��: not found
./test-pio.bin: line 3: @��PE�@��0������/�V����@F���Fa��: not found
./test-pio.bin: line 4: @��PE�@��0������/�V����p@���/��u0�u0�u0�u0�?: not found
./test-pio.bin: line 1: ��@��P����P�: not found
./test-pio.bin: line 1: ��@��P��U�����p�����G: not found
./test-pio.bin: line 1: P�p��V�0�����������$�0�MbP�: not found
./test-pio.bin: line 1: ��q�: not found
./test-pio.bin: line 2: can't open ��
����                                0����4���@��
     ����^�0���0��
                     0�����0��
                               0�����㼟���ȟ��묟�V�0��P�: no such file
./test-pio.bin: line 2: ���P��t�L���0��S����P��P���
                                                               0����: not found
./test-pio.bin: line 3: ���P����
                                 0�@��0���㌟�
                                                0�@���넟��봟����ᬟ����᤟��렟��뜟���������7�PP��������,��2�0��P�������j0�j0k0Pv0: not found
./test-pio.bin: line 2: P�
                           ����/����/�������: not found
./test-pio.bin: line 5: can't open �
                                     0������0��
                                                0�������5����3����1�0�����0����
                                                                                      0����: no such file
./test-pio.bin: line 5: R�: not found
./test-pio.bin: line 3: 0��P�2���/����/��@�: not found
./test-pio.bin: line 3: C���0��,0��������: not found
./test-pio.bin: line 23: p����p������Q�P���������������A-造��
                                                                �����P0�@��L0���C��%à��
                                                                                       �}/�0��#�: not found
./test-pio.bin: line 28: r�����U�: not found
./test-pio.bin: line 1: syntax error: unterminated quoted string

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
```
Any suggestions any suggestions are welcome

TonyHan11, 2020-12-03:
The binaries built from the software package are baremetal code and are not suitable for running on any OS.

pragnesh036, 2020-12-03:
so how can i execute my  test-pio example available at atmel-software-package-master/examples/test-pio/ to my SAMA5D3 board

pragnesh036, 2020-12-03:
> The binaries built from the software package are baremetal code and are not suitable for running on any OS.

so how can i execute my test-pio example available at atmel-software-package-master/examples/test-pio/ to my SAMA5D3 board

TonyHan11, 2020-12-03:
The examples could be run by GNU GCC or IAR EWARM (details in [README.md](https://github.com/atmelcorp/atmel-software-package/blob/master/README.md)). It is also OK to put the binary to a boot media and boot the board from the media.