atmelcorp/atmel-software-package #109: SPI flash programming much slower than necessary

tanstafel, 2021-02-05:
SPI flash programming is done page wise and after a page is written a status bit is checked to wait for completionl. Depending on flash this may take some 100µs or maybe even faster. The code in `drivers/nvm/spi-nor/spi-flash.c` waits at least a millisecond. If you have to write 16MB or more this can sum up to a large overhead. I propose to change the wait loop to check not every ms but every µs. This might be overkill but there is also no real disadvantage checking a bit more often than necessary. Proposed patch:

```
--- a/drivers/nvm/spi-nor/spi-flash.c
+++ b/drivers/nvm/spi-nor/spi-flash.c
@@ -97,8 +97,8 @@ static int spi_flash_is_ready(struct spi_flash *flash)
 
 int spi_flash_wait_till_ready_timeout(struct spi_flash *flash, unsigned long timeout)
 {
-       unsigned long delay = 1L; /* 1ms */
-       unsigned long loop = (timeout + delay - 1) / (delay);
+       unsigned long delay = 1L; /* 1us */
+       unsigned long loop = (timeout * 1000 + delay - 1) / delay;
        int rc;
 
        if (!loop)
@@ -110,7 +110,7 @@ int spi_flash_wait_till_ready_timeout(struct spi_flash *flash, unsigned long tim
                if (rc)
                        return 0;
 
-               msleep(delay);
+               usleep(delay);
        }
 
        return -ETIMEDOUT;

```

tanstafel, 2021-02-05:
Without the patch I need ~176 seconds to erase and write 13MB. With the patch this reduces to ~128 seconds.

TonyHan11, 2021-02-07:
Thank you for your suggestion and patch, would you like to push your patch and [create a pull request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) to the master branch? Thanks.

closed 2021-02-12 by tanstafel