atmelcorp/atmel-software-package #114: Correct twid NACK management in polling mode

hilt0n, 2022-05-13:
I'm having trouble with TWI NACK detection in master mode. While I should receive a NACK error (Write/Read to inexistent device on the bus), I receive a RX TIMEOUT because NACK seems to be not correctly handled, at least on the ATSAMA5D44.

Following the datasheet, the SR_NACK bit appears in case of NACK from device once the TXCOMP bit is set.
It means checking this bit before having sent a STOP conditions will always return false.

It also means, during a transfer, the NACK condition will unfortunately only be detected at end of the transfer even if NACK is received during device addressing (this is what I understand, maybe I'm wrong).

Still, I have been able to fix the issue (to confirm my supposition) by replacing the following code (drivers/i2c/twid.c:514-516):
```
/* Before receive last data, send stop */
if (desc->flags & BUS_I2C_BUF_ATTR_STOP){
	twi_send_stop_condition(addr);
	if (_check_nack(desc))
		return -ECONNABORTED;
}
```

It clear the intention is to manage it since there is `_check_nack()` at some places but visibly, this doesn't work for the ATSAMA5D44.

hilt0n, 2022-05-13:
Well, it was working by chance because I had printf around..

Seems logic in fact that if the SR_NACK is set only on STOP condition, we will get RX TIMEOUT before getting NACK regarding the current implementation.

I keep it open in case you have a solution to fix this issue.

hilt0n, 2022-05-13:
The problems comes from the fact reading SR clear the NACK flag. While SR_NACK is checked all around, it's most likely always 0 because SR is read by other commands (_check_tx_timeout, etc)

TonyHan11, 2022-05-18:
In case of write/read to inexistent device on the bus , the NACK bit will be set at the same time when TXCOMP bit is set.

About the STOP condition, according to `46.6.3.4 Master Transmitter Mode` in the [datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/MPU32/ProductDocuments/DataSheets/SAMA5D2-Series-Data-sheet-ds60001476G.pdf): `If the slave does not acknowledge the byte, then the Not Acknowledge flag (NACK) is set in the TWIHS Status Register (TWIHS_SR) of the master and a STOP condition is sent.` For TWI in FLEXCOM it is a bit different: the auto-STOP can be disabled by set FLEX_TWI_MMR.NOAP (No Auto-Stop on NACK Error) to “1”.

The status of NACK can be updated in the status register the first time TXCOMP bit changes from 0 to 1. Please notice that this bit has the property “cleared on read”.

In softpack NACK = 1 is omitted and not processed, . Users may add code to remember the status of NACK bit where TWI_SR is read (e.g `twi_get_status()`)  or enable interrupt triggered by NACK bit and process it.

hilt0n, 2022-05-18:
Thanks for the update, it's indeed what I see, good to know that the desired behaviour.

closed 2022-05-18 by hilt0n