atmelcorp/atmel-software-package #115: Where to find recent SAM-BA sources?

zougloub, 2022-08-01:
Sorry for the out-of-scope issue but https://github.com/atmelcorp/sam-ba is not open for issues anymore so I found a closely related package to report on.

@noglitch in https://github.com/atmelcorp/sam-ba/commit/de00221aae35b42d3808cf481b764cc1087ccf50 you indicate that SAM-BA can be downloaded "from [Microchip website](http://www.microchip.com)", however while I could get the compiled binaries [here](https://www.microchip.com/en-us/development-tool/SAM-BA-In-system-Programmer), I can't seem to be able to find the (GPL) corresponding sources from anywhere... please advise.

Best regards,

noglitch, 2022-08-17:
Hi,
The latest source code that you can have for sam-ba is the one published on [https://github.com/atmelcorp/sam-ba/](https://github.com/atmelcorp/sam-ba/)
and is there: https://github.com/atmelcorp/sam-ba/releases/tag/v3.3.1 (see source code links).
Regards,
  Nicolas

DrFurunkel, 2023-01-12:
Hi guys,

sorry for hijacking...
We work with a QSPI-NAND on the x60. Works perfect. 
For ISP, i need to write the qspi-nand Applet. 
Can i use the https://github.com/atmelcorp/atmel-software-package as source for my applet or does the recent SAM-BA Releases (Qt-based) use Applets that aren't compatible to the atmel-software-package generated ones?

Urs