atmelcorp/atmel-software-package #116: More than one CAN instances does not work

frebjo, 2022-09-21:
In e.g. version b32c2808c53cd53534c76367a2481819f4ee3a2d of the https://github.com/atmelcorp/atmel-software-package.git
in the file atmel-software-package/drivers/can/mcand.c
The global variable

static struct _cand_ram_item mcan_ram_item[MSG_RAM_ITEM_CNT];

needs to be unique for each CAN interface
thus instead

static struct _cand_ram_item mcan_ram_item[CAN_IFACE_COUNT][MSG_RAM_ITEM_CNT]; 

When CAN interfaces are using the same buffer data will overwritten by the other interface.
On line 1214, 1215 change 
assert(index <= ARRAY_SIZE(mcan_ram_item));
desc->ram_item = mcan_ram_item;

to 

 desc->ram_item = mcan_ram_item[mcan_get_index(mcan)];

TonyHan11, 2022-09-23:
Hi,
Thank you for your report. Would you like to create a pull request for this issue? Or I can update it if you don't mind.

frebjo, 2022-09-23:
Hi!It is ok if you update it.Best regards Fredrik 

Skickat från Yahoo Mail på Android 
 
  På fre, sep.

TonyHan11, 2022-09-26:
Hi,
I've create a pull request to fix this issue, would you like to review it? Thank you very much.

frebjo, 2022-09-26:
Hi!I have reviewed your pull request and it is OK for me.Best regards Fredrik 

 
 
  På mån, sep.

TonyHan11, 2022-09-27:
Thank you for your feedback and I'll merge it to the master branch later.

closed 2022-09-28 by TonyHan11